JMRI/JMRI #2783: 4.7.* fails any panel with linkinglabel construct

KenC57, 2016-12-29:
It seems the item linkinglabel is not in the validate for panel xml files. The error given is:

`2016-12-29 18:11:38,321 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/kcameron/JMRI/PanelPro/LinksCliffA.xml Exception: org.jdom2.input.JDOMParseException: Error on line 37: cvc-complex-type.2.4.a: Invalid content was found starting with element 'linkinglabel'. One of '{icons, positionablelabel, positionablepoint, positionableRectangle, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, multisensoricon, fastclock, locoicon, indicatortrackicon, indicatorturnouticon, PortalIcon}' is expected.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 37: cvc-complex-type.2.4.a: Invalid content was found starting with element 'linkinglabel'. One of '{icons, positionablelabel, positionablepoint, positionableRectangle, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, multisensoricon, fastclock, locoicon, indicatortrackicon, indicatorturnouticon, PortalIcon}' is expected.
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)
	`

Cliff Anderson found it and posted the following panel file. It has only internal connections needed so the test is really simple.

https://groups.yahoo.com/neo/groups/jmriusers/files/ProblemsBeingWorkedOn/Cliff%20Anderson/

-ken c

KenC57, 2016-12-30:
This is going to take someone who understands DTD files. I'm not the one for that stuff. I get the concept but that's as far as I've been with those, using them, not writing them. But a quick grep shows none of them know what a linkinglabel is for. It extends the positionableLabel.

Pugwash1, 2016-12-30:
Hi
OK, I can read the panel, but when I try and click on things I get this error " ERROR - Error handling link [AWT-EventQueue-0]
java.net.MalformedURLException: no protocol: panel/ControlPanel/ControlPanel1%20Editor". is this likely due to implied root or that its a windows specific link.

KenC57, 2016-12-30:
I'm finding the panel editor links work, but not the control panel editor
links. I've sent message to Cliff to see if he has more detail or can check
his work.

 

I haven't used the feature yet, but have two layouts that need it.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2016-12-30:
LinkingLabels must always have a URL that starts with ```frame:``` or can be used to construct a  [java.net.URL][1] object. This is enforced in the [doMouseClicked() method][2] of LinkingLabel.

[1]: https://docs.oracle.com/javase/8/docs/api/java/net/URL.html
[2]: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/display/LinkingLabel.java#L64

bobjacobsen, 2016-12-30:
> (@KenC57) I'm finding the panel editor links work, but not the control panel editor links. I've sent message to Cliff to see if he has more detail or can check his work.

Can we get a sample file for control panel editor, please? Cliff's file (referenced above from the jmriusers file area) uses paneleditor.

KenC57, 2016-12-30:
His example file seems to have one or more of every type of pane. Only issue
right now is if all the URL entries are valid or not. Looking at the xml it
seems there's a difference between the format for the different editor
panes. I'm not sure what is the right fix for some of the syntax.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-12-30:
Thanks. Just figured out that control panel editor isn't actually mapping elements to classes in the usual way.  It works, so no need to change it now, but we might want to look at that in the future.

Meanwhile, here are the url elements in that file:
```
      <url>/panel</url>
      <url>frame:PanelThing2</url>
      <url>panel/ControlPanel/ControlPanel1%20Editor</url>
      <url>panel/ControlPanel/ControlPanel2%20Editor</url>
      <url>frame:LayoutPanel1</url>
      <url>/panel</url>
      <url>frame:PanelThing1</url>
      <url>panel/ControlPanel/ControlPanel1%20Editor</url>
      <url>panel/ControlPanel/ControlPanel2%20Editor</url>
      <url>frame:LayoutPanel1</url>
      <url>/panel</url>
      <url>frame:PanelThing1</url>
      <url>frame:PanelThing2</url>
      <url>panel/ControlPanel/ControlPanel2%20Editor</url>
      <url>frame:LayoutPanel1</url>
      <url>/panel</url>
      <url>frame:PanelThing1</url>
      <url>frame:PanelThing2</url>
      <url>panel/ControlPanel/ControlPanel1%20Editor</url>
      <url>frame:LayoutPanel1</url>
      <url>/panel</url>
      <url>frame:PanelThing1</url>
      <url>frame:PanelThing2</url>
      <url>panel/ControlPanel/ControlPanel1%20Editor</url>
      <url>panel/ControlPanel/ControlPanel2%20Editor</url>
```

It currently loads, but trying to click to the "ControlPanel1 Editor" window from "LayoutPanel1" fails with:
```
    [java] java.net.MalformedURLException: no protocol: panel/ControlPanel/ControlPanel1%20Editor
```

adding "frame:" to the front of the URL gives a different error:
```
display.LinkingLabel                  ERROR - Frame 'panel/ControlPanel/ControlPanel1%20Editor' not found, cannot link to it. [AWT-EventQueue-0]
```

Since that might be due to the "%20", I replaced that with a space and still got something similar:
```
display.LinkingLabel                  ERROR - Frame 'panel/ControlPanel/ControlPanel1 Editor' not found, cannot link to it. [AWT-EventQueue-0]
```

The links to `frame:PanelThing1` and `frame:PanelThing2` work fine.  The link to `/panel` (or `frame:/panel`) doesn't work.

On the off change the problem with the first one is the full path name, try `frame:ControlPanel1  Editor`: works!! For completion, also tried `frame:ControlPanel1 %20Editor`
```
display.LinkingLabel                  ERROR - Frame 'ControlPanel1%20Editor' not found, cannot link to it.
```

These were all tests of links from Layout Editor panels, and perhaps somebody will try others. But from this I conclude:
- The `<url>` element needs to start with "frame:" currently.
- The name that follows the `frame:` should be the title, as shown, of a panel window.  
   - Escapes in the URL are not processed.  
   - There's no pathname to traverse.

I don't know what, if anything, of this is new.  Nor do I know how many existing files are out there that don't work.

If those really are the rules, then we can write a schema to validate to that.  That's easy.  But I suspect that those should _not_ be the rules, if this file used to work.  And I don't know what change(s) would have broken it beyond validation.

rhwood, 2016-12-30:
Looking at the URLs, it appears the URLs that start with "panel/" only work in the web server, since they have the form that it wants. The OP's club makes extensive use of the web server to display the panels, so it may be that they never worked in JMRI, but the values were simply passed to the web server where they worked in a browser.

KenC57, 2016-12-30:
Randal,

 

Is the correct behavior that it should work on JMRI and that the web will
translate??

 

Also from poking around it seems there isn't a direct item in the editors
for this, but some sort of create positionable label and then edit the xml??
Any clicks I've been trying don't seem to get me anywhere to change the
links, only the text of the label.

 

-ken c

KenC57, 2016-12-30:
Here is a fixed up copy of the file. All converted to some frame: construct.
I also exited the 'editor' mode on the control panel editor panes and thing
are ok now. No faults now.

-ken c

rhwood, 2016-12-30:
The web server accepts either form. I don’t know what the correct behavior in JMRI is, or if the behavior in JMRI is deliberately different for the different kinds of panels.

In the Control Panel Editor, its an “Icon” element, and is editable using the “Edit Link” menu item. In the Layout Panel Editor or a Panel Editor, I don’t know; I didn’t see anything obvious. (To be completely clear, I only helped code the web server, and really don’t understand the the panel code.)

Randall Wood

KenC57, 2016-12-30:
Randall,

 

After cleaning them up to all use the frame:, its' working nicely. It
changes the focus on JMRI and it replaces the current one when in the web.
So I'd say that's working nice.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2017-01-28:
The original file on Yahoo is gone (why do we keep deleting 1-2MB files "due to press of space" when we're using 80MB of 2GB?), and I don't see any other attached files.  If anybody has a test case that we can include, please open a new issue.  In the meantime, I think this is done, so am closing it.

closed 2017-01-28 by bobjacobsen

KenC57, 2017-01-29:
I just checked the current HEAD against the only test files I had that used
linkinglabel and it passes fine. Wasn't there a file named
LinksCliff[AB].xml attached to the issue??

 

-ken c

Cliff-Anderson, 2017-01-29:
Two exact copies of my original file, that only works for Windows and does not include the icon form of a linkinglabel has been included in the the github directories:
C:\Users\cliffaa-rr\git\JMRI\java\test\jmri\jmrit\display\verify\LinkingLabelPanelEditor.xmlC:\Users\cliffaa-rr\git\JMRI\target\classes\jmri\jmrit\display\verify\LinkingLabelPanelEditor.xml
Both timetaged 1/22/2017 7:26 PM
The "fix" for non-windows, as described by Ken, was to remove the "%20Editor" and convert the two Control Panel created panels to appear as Panel created panels.  As far as I can tell, this appears to be a one-way conversion as the only way to revert back to Control Panel editing was by way of the XML editor.  Linking Labels are only visible with the Control Panel editor, so access is constrained.  
A revised version based on Ken's instructions is attached. 
Cliff Anderson