JMRI/JMRI #2786: Unwanted creation of Internal Turnouts when creating LocoNet Turnouts

HaltonSub, 2016-12-30:
When I create a LocoNet turnout (system name: LTxxx),  providing a unique user name, the program creates a Internal Turnout (ITLTxxx) and assigns the same unique user name.   This causes problems as Logix statements seem to choose the User name associated with the IT.    Also the turnout table associated with creating a new PanelPro turnout icon does not show the LTxxx but only the ITLTxxx.

Using 4.2, the system seemed to associate automatically the LT and IT turnouts, but with the upgrade to 4.6, there is no longer any association.

I have worked around the issue by manually changing the ITLTxxx User name, which then updates the turnout variables used in Logix to the new ITLTxxx User Name.    With that I have had to go through all of the Logix conditionals make sure that all Turnout variables are then changed back to the User Name associated with the LTxxx.

Thanks ... Dave M

KenC57, 2016-12-30:
Dave M,

 

When creating a turnout, the prefix is supplied by the window showing
'Internal' or 'LocoNet'. So you should just be entering the number 'xxx' for
the system name, it adds the 'LT'. Same for Internal, it will add the 'IT'
prefix.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

HaltonSub, 2016-12-30:
Hi Ken

Thanks for your response

I am sure that I created the LTxxx as you described, otherwise one sees the error message "Unable to convert xxx to a valid Hardware Address"

What happened is that it seems to have created ITLTxxx turnouts on its own.

However I just tried to create a new loconet turnout using just xxx, and the system created the LTxxx as expected.    I then saved, restarted, and there was not an ITLTxxx created.

???

Dave

KenC57, 2016-12-30:
If the pulldown had 'Internal' showing, it would accept the 'LTxxx' as a
system name and make it 'ITLTxxx'. That's my guess, sometimes you didn't
notice which it had popped up with. It's easy not to notice.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

HaltonSub, 2016-12-30:
good thought but I would have noticed that the LTxxx turnout designation missing

also, a while ago, I actually deleted all of the “duplicate” IT turnouts, only to see them reappear 

Dave

HaltonSub, 2017-01-03:
Hi Ken

Creation of IT for every LT has appeared again

We have a config where we have a “development” pc and a “production” pc.

On the Dev side, the program works and I had deleted all unneeded phantom ITs .   All was well.

But when I transferred the file to the Prod pc, once again the phantom ITs appeared and caused all sorts of issues as Logix and Routes chose the IT instead of the LT

Dave

btw… did I meet you two years ago at the Springfield train show ?  … we went out to dinner with Dick and Karen Bronson

KenC57, 2017-01-03:
Dave,

 

Yes that would have been me.

 

One question I have is do you use only systemNames or usernames?? Next
question is if you have 'fully qualified' systemNames?? Here I mean they are
all listed as LTsomething or ITsomething??

 

The latter is because JMRI used to support having the '123' and it would use
the default system. But in today's world it wants explicit LT123 if you mean
on the LocoNet.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

HaltonSub, 2017-01-04:
Hi Ken

All Turnouts are as you classify ‘fully qualified’.   Every turnout has both a system name and user name.

I have used JMRI for many years having programmed the former Model Railroad Club of Toronto O scale layout which featured full CTC and automatic cab routing.   It had over 60K lines of code and worked very well.

Dave

HaltonSub, 2017-01-04:
reading this again you asked a few questions but i only answered the second

answer to first question

typically I use the user name when setting up the conditionals in logix

Dave

KenC57, 2017-01-04:
Dave,

 

So far you're case isn't matching up with any prior cases that I know of.
We've had issues with wrong connection prefix being used but that was always
when old code was not using 'fully qualified' system names.

 

I'm thinking the panels and any scripts may need to be posted for the issue
to be found. So far I'm not clear on what it could be.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

KenC57, 2017-01-22:
The case is in pre 4.7.1 versions, we still load files when we have a failure of a port connection. So my test case was a LocoNet system and when the error messages finished being posted, I had a bunch of ITLT* and ITLS* elements in the tables. Since it didn't have a L connection to use, it passed all those elements to the Internal connection. If a user saves that panel, they lost all the original LocoNet connections.

But as of 4.7.1ish, we stop the startup preferences from running if we have a failure during the port creation phase. HOWEVER, if you try loading a panel into a running session that does not have the matching connection prefix, you get the ITLT* type results!! So this begs the question:

"Should a panel load prevent loading if the matching connection prefix does not exist??"
-ken c

KenC57, 2017-01-22:
One further thought. On the plus side this behavior doesn't lose all the usernames for the entries and keeps their existance around. You generally get a number of missing beans, I'm guessing from where things were using system names which of course don't exist.

But on the good side, a recovery script could be made that would recover the bad file by doing moves of all the user names from the ITLT back to LT matching entries, same for the ISLS entries. So it might help out the cases of users who overwrite the same file etc... But on cases where something was still looking for the LT name, and didn't find it during the load, I don't know what it does, I'm guessing nulls.
-ken c

bobjacobsen, 2017-01-22:
@KenC57 "Should a panel load prevent loading if the matching connection prefix does not exist??"

Perhaps we should unpack this a bit first.  (There are questions at the end)  

Originally, turnouts and sensors where referenced by names like "23" and "4".  That was very convenient for users, because having put together their hardware, they generally knew those numbers and had some control over them. The `TurnoutManager#provideTurnout(..)` and `SensorManager#provideSensor(..)` handled all the behind-the-scenes, hardware-specific stuff for you.  You didn't have to create things in a table first, you just referenced them.  (Originally, the tables didn't even exist in the UI)

Once multiple systems came along, including "internal" (virtual) systems, the system name became more important.  You could still refer to Sensor "23" and get the default, but you could also refer to "LS23".  (Note that even now, if you go to the Turnout Control and type "4" in the turnout field, it'll happily operate device "4" on the default system; you don't need to specify LT4 or NT4 or whatever; I think that's the _right_ behavior)

When storing these, the underlying type is defined in two ways:
- The XML references the type through a `class` attribute on the enclosing `<turnouts>` element.
- The individual items have a system name stored with them that has a prefix.

When loading them, the ConfigXML classes have code to do system specific things, but they usually defer back to an abstract base class for common parts (this is also the right behavior).  For example, [loconet.configurexml.LnTurnoutManagerXml](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/configurexml/LnTurnoutManagerXml.java) doesn't have any system-specific stuff to do, so it just defers back to the `loadTurnouts(..)` method of [jmri.managers.configurexml.AbstractTurnoutManagerConfigXML](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/managers/configurexml/AbstractTurnoutManagerConfigXML.java).  That, in relevant part, creates the Turnout and sets names via:
```
            Turnout t = tm.getBySystemName(sysName);
            if (t == null) {
                t = tm.newTurnout(sysName, userName);
            } else if (userName != null) {
                t.setUserName(userName);
            }
```

Notice an interesting thing:  There's nothing here that requires the type to be the same!  If the panel file has a `<turnouts>` element with a `class` method specifying a `jmrix.loconet` class but the system prefix on the system name is for NCE, well, that'll usually work fine. (There may be systems that have system-specific processing after the general code that might cause trouble, but I can't think of one right now).

Is that bad?  I don't know.  If it is, it wouldn't be that difficult to add a test in the specific ConfigXML class to check for the proper system connections before invoking the common processing.  But coding a test for consistency of _system_ _names_ might be difficult, because there can be multiple ones.   And I'm not sure this is a bad thing, because I can think of times when it would help with migrations.  (E.g. "I have this huge layout with system A using e.g. ALnnn, and I'm switching to system Z; I'll just set my system prefix to 'A' for my Z connection and import my existing files!")

This is made a bit more tricky by a special case in system prefix handling for the internal system:  It accepts free-names, not just numbers.  This sort-of made sense back when user names were less commonly used:  Why not have an internal turnout names "ITLunchtime" to remember some internal state?  There's no need to correspond to any real hardware, no problems like "I don't know what DCC address turnout 'Lunchtime' would map onto".  The net effect of this is if you say 
```
InternalTurnoutManager.provideTurnout("Lunchtime");
```
you get a Turnout with system name "ITLunchtime" and no error.  

So the code thinks that 
```
InternalTurnoutManager.provideTurnout("LT12");
```
is a perfectly reasonable request, and give you a Turnout with system name "ITLT12".

This all evolved for very specific reasons, and when operated correctly works very well.  But it doesn't hurt to reconsider things occasionally.  So the questions become:

- Systems don't enforce their system prefixes while loading.  Should they?

- The Internal system doesn't enforce any kind of system prefix, during normal operation nor during loading.  Should it?

My personal opinion right now, subject to change, is that we should _not_ change this just because somebody did something broken.  Sometimes improper behavior results in improper results.  The user that prompted this discussion bypassed _multiple_ warnings, and got unexpected behavior.  Such is life.

bobjacobsen, 2017-01-22:
@KenC57 
> But on the good side, a recovery script could be made that would recover the bad file by doing moves of all the user names from the ITLT back to LT matching entries, same for the ISLS entries. So it [...]

If somebody wants to write that, they're of course welcome to.  It's formally impossible to get completely right (it's isomorphic to the [Halting Problem](https://en.wikipedia.org/wiki/Halting_problem)), but you might be able to go a long way toward making something useful.

But how about just recommending backups? 

At some point, entropy is going to win.  It's simply not possible to anticipate a way to recover from every wrong way of operating a computer program.  And the more we promise to help people who going that direction, the more is going to be expected.

KenC57, 2017-03-06:
Dave M,

You started this case and it has traveled a bit. Going back to the original issue, did it happen immediately or only after doing a 'save/restart' for JMRI?? I'm guessing at if the ITsomethings were only occurring during the system startup or was it happening before the save?? Best test for this is if the values of ITLT* show up after the save, but before loading it in JMRI. The issue of a failed connection is cured since you can't load a file without the matching connections. That case was causing a number of cases of LT* turning into ITLT* entries.

KenC57, 2017-03-12:
At this point (just prior to 4.7.2), if a connection fails to get created right, it won't load the panel. But if a panel is being loaded where there isn't a matching connection, you get conversion of real system names to internal system names.

HaltonSub, 2017-03-17:
Hi Ken

Since we invoked the new Turnout State Persistence script the problem has gone away.

My recollection of the generation of the phantom turnouts (ITLTxxx) was under the following conditions:

1) Use of original Turnout State Persistence script

2) Modification of panel file was done on our Test PC, sometimes resulting in modifications to the Turnout Table

3) File transferred to Production PC

Dave

geowar1, 2017-10-30:
Can this be closed?

closed 2020-01-13 by bobjacobsen