JMRI/JMRI #2791: Programmer types ESU and Zimo are not valid.

rhwood, 2016-12-30:
The programmer files ```xml/programers/ESU.xml``` and ```xml/programmers/Zimo.xml``` are invalid. Tests for these two programmer files are ignored in ```jmri.jmrit.symbolicprog.tabbedframe.CheckProgrammerNamesTest``` due to this. Tests for other files in ```xml/programmers``` are enabled in #2769.

dheap, 2016-12-30:
They are both so ancient I wonder if they are any use to anyone anymore!

>The programmer files xml/programers/ESU.xml and xml/programmers/Zimo.xml are invalid.

bobjacobsen, 2016-12-31:
They do pass schema checks, but the CheckProgrammerNamesTest check also checks the variable names in each programmer file against the list of standard names in `names.xml`.  That's what's failing, with lots of messages like:
```
Variable not found in name list: name="Flash Light Output 1"
Variable not found in name list: name="Flash Light Output 2"
Variable not found in name list: name="Flash Light Output 3"
Variable not found in name list: name="Flash Light Output 4"
Variable not found in name list: name="Flash Light Output 5"
Variable not found in name list: name="Flash Light Output 6"
Variable not found in name list: name="Flash Light Inverted Output 3"
Variable not found in name list: name="Flash Light Inverted Output 4"
Variable not found in name list: name="Flash Light Inverted Output 6"
```

The purpose of the names.xml test is to make sure that decoder files and programmer files match up:  Programmer files, particularly Comprehensive and Advanced, should show what's in the decoder file.  

The ones showing in the list above aren't in Comprehensive and Advanced, but they are in Zimo (and also CT Elektronik) definitions:
```
% grep -r "Flash Light Output 2" xml/decoders/
xml/decoders//CT_Elektronik_DCX_30_V.xml:      <variable item="Flash Light Output 2" CV="62" mask="XXXXXXVX" default="0">
xml/decoders//CT_Elektronik_DCX_30_V.xml:        <label>Flash Light Output 2</label>
xml/decoders//CT_Elektronik_DCX_30_V.xml:      <display item="Flash Light Output 2" format="checkbox" label="Output 2">
xml/decoders//Zimo_MX61_1.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX61_1.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX61_2.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX61_2.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX62_1.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX62_1.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX62_64_1.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX62_64_1.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX64(H)_1.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX64(H)_1.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX66-2000-11.xml:      <variable item="Flash Light Output 2" CV="118" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX66-2000-11.xml:        <label>Flash Light Output 2</label>
xml/decoders//Zimo_MX68(L)_1.xml:      <variable item="Flash Light Output 2" CV="63" mask="XXXXXXVX" default="0">
xml/decoders//Zimo_MX68(L)_1.xml:        <label>Flash Light Output 2</label>
```

The CT Elektronik files handle this by having panes locally, within the definition. That way they work with Comprehensive and advanced. 

But the Zimo files rely on using the Zimo programmer file to display them. So, at least for these variables in those decoder files, the Zimo programmer file isn't obsolete.

The question is what to do about this in the long term.  We could rework the Zimo programmer files to work with Comprehensive and Advanced, which might take adding some more display elements to those (and if so adding those to names.xml).  That's a lot of work, but given that Zimo decoders are complex, popular and likely to be around, this might be worthwhile.

Or we could leave this as it is.

dheap, 2016-12-31:
So it appears that users who select these definitions don't get the correct display unless they manually do Preferences->Roster->Programmer and select Format: Zimo and Save.
This:
- Messes up the pane display for other brands of decoders.
- Doesn't work from the New Loco window anyway as this always forces the Comprehensive programmer to open.

So they are fairly broken as-is.

Given that we now have [Pane Replacement](http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#pane_replacement) we can fairly easily move `Zimo.xml` to `xml/decoders/zimo/`, setup up an `xi:include` for the affected files and make sure pane names in `Zimo.xml` match those they are intended to replace.

Some work, but not a massive amount.

bobjacobsen, 2016-12-31:
> So it appears that users who select these definitions don't get the correct display unless they manually do Preferences->Roster->Programmer and select Format: Zimo and Save.

When I read that, I thought that there was a simpler way, where you selected the programmer as you opened the decoder.  But the new Roster Frame (e.g. DP3 version) doesn't seem to have that; I was probably remembering the older interface, which did (see e.g. [this image](http://jmri.sourceforge.net/help/en/manual/DecoderPro3/images/Set_Up.png)).

Without a way to select them easily, it's not clear what value alternative programmer definitions really have.  In addition to using Pane Replacement to (eventually) replace the Zimo and ESU programmers, should  we also make choice-of-programmer more visible in the interface?

rhwood, 2018-03-16:
OP is no longer concerned.

dheap, 2018-03-16:
I think it's time we retired these. I know of no ESU files that rely on the ESU one and having it is confusing.

@bobjacobsen Are you sure the ZIMO one is still needed?

bobjacobsen, 2018-03-16:
@dheap Will check the Zimo one.

bobjacobsen, 2018-03-16:
@dheap Sorry, autocorrect: I'll check the Zimo one.

bobjacobsen, 2018-03-16:
There are some Zimo decoders with non-standard variables that reference names in the Zimo programmer:

xml/decoders//Zimo_MX68(L)_1.xml
xml/decoders//Zimo_MX62_64_1.xml
xml/decoders//Zimo_MX66-2000-11.xml
xml/decoders//Zimo_MX61_2.xml

and probably others. If we want to get rid of the Zimo programmer, we could move some of those to standard names and extra panes in Advanced.

closed 2021-04-14 by stale[bot]