JMRI/JMRI #2793: Unable to delete conditionals or logix

HaltonSub, 2016-12-31:
Hi Again...   another issue I've encountered is that I am unable to delete any conditionals or logix themselves.     Dave M

KenC57, 2016-12-31:
Dave M,
Which version are you using?? I've been using the HEAD and 4.6 and not having issues with Logix.
-ken c

HaltonSub, 2016-12-31:
We are using 4.6.    What do you mean by HEAD?

KenC57, 2016-12-31:
Sorry, HEAD is what the dev group refers to 'latest code checked in', so it
would be 4.7.1 when that gets released. Do you get any messages about it
failing or it acts like it deleted, but when you view the list of
conditionals, you still see it??

-ken c

HaltonSub, 2016-12-31:
Hi Ken

When I try to delete a conditional, the menu appears but when clicked, nothing happens.   

When I try to delete a logix, again the menu appears, but it does not allow it to be deleted, and it sometimes freezes and can only be unfrozen by closing the logix table view.

There is no error message.

This was happening before we upgraded to 4.6 about a week ago

Dave

dsand47, 2017-10-30:
I have re-created the problem.  It deals with removing listeners during the delete process.  It was fixed as of 4.7.2.

This issue can be closed.

Dave Sand

closed 2017-10-30 by bobjacobsen