JMRI/JMRI #2797: Does awt_multiclick_time make any sense?

bobjacobsen, 2017-01-01:
`roster.RosterFrame` has the following code:

```
        try {
            clickDelay = ((Integer) Toolkit.getDefaultToolkit().getDesktopProperty("awt.multiClickInterval"));
        } catch (RuntimeException e) {
            try {
                clickDelay = ((Integer) Toolkit.getDefaultToolkit().getDesktopProperty("awt_multiclick_time"));
            } catch (RuntimeException ex) {
                clickDelay = 500;
                log.warn("Unable to get the double click speed, Using JMRI default of half a second: " + e.toString());
            }
        }
```

@pabender recently changed the log level from ERROR to WARN, and I'd like to drop it further to INFO (or perhaps even DEBUG) to try to reduce the number of "is this a problem?" emails.  

While looking at it, I started to wonder whether the inner try/catch ever works for anybody, in the sense of "get(ting) the double click speed" from `awt_multiclick_time`.  I can't find _any_ online references to `awt_multiclick_time` as a valid DesktopProperty.  Does that inner try/catch always drop through and  take the JMRI default?

silverailscolo, 2017-01-01:
I always got/get this warning in the Console on macOS, regardless of Locale.

rhwood, 2017-01-03:
The property should probably be "awt.multiClickInterval" as suggested by line 1469 of http://www.docjar.com/html/api/sun/awt/X11/XToolkit.java.html

rhwood, 2017-01-03:
The Java tutorial suggests we should merely pay attention to the click count in mouse events. I can't find any references to awt.multiClickInterval in the Java 8 sources, so I'm wondering if these properties once were a thing that we can't rely on now.

PR #2901 (2017-01-21): Remove awt_multiclick_time when setting double click speed
Removes the reference to `awt_multiclick_time` when trying to set the double-click speed.  

Lowers the messages from that to DEBUG level (from ERROR and WARNING) since they Happen All The Time. 

Closes #2797 when merged.
closed 2017-01-21 by bobjacobsen