JMRI/JMRI #2805: DecoderPro shows Programmers online with invalid configuration

rhwood, 2017-01-02:
I have a profile with an invalid connection (LocoNet LocoBuffer-USB with port name /dev/cu.Non-Existant-Port).

In 4.7.1-ish build from JMRI/master, although errors are shown at startup, and the connection is correctly red in the Preferences window, the roster window shows "Service Mode Programmer is Online" and "Operations Mode Programmer is Online" in green.

In 4.6, the roster window correctly shows the programmers are offline in red.

pabender, 2017-01-02:
Sent from my iPhone

> On Jan 2, 2017, at 6:57 AM, Randall Wood <notifications@github.com> wrote:
> 
> I have a profile with an invalid connection (LocoNet LocoBuffer-USB with port name /dev/cu.Non-Existant-Port).
> 
> In 4.7.1-ish build from JMRI/master, although errors are shown at startup, and the connection is correctly red in the Preferences window, the roster window shows "Service Mode Programmer is Online" and "Operations Mode Programmer is Online" in green.
> 
> In 4.6, the roster window correctly shows the programmers are offline in red.
> 
This is going to be as a result of the code Input in place last week in the roster frame.

What I suspect you are seeing here is the result of updating the status based on the port name without also specifying the connection name ( I.e. Using the old form, not the new form ).  That was designed to work ( and the tests shows it does ) but I may have missed a test case.

I can look into that more tomorrow.

Paul

rhwood, 2017-01-03:
The RosterFrame also crashes during construction if the connection is "None":
```
2017-01-02 20:51:34,727 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.roster.swing.RosterFrame.updateProgrammerStatus(RosterFrame.java:1534)
	at jmri.jmrit.roster.swing.RosterFrame.buildWindow(RosterFrame.java:364)
	at jmri.jmrit.roster.swing.RosterFrame.<init>(RosterFrame.java:141)
	at jmri.jmrit.roster.swing.RosterFrame.<init>(RosterFrame.java:132)
	at jmri.jmrit.roster.swing.RosterFrame.<init>(RosterFrame.java:128)
	at jmri.jmrit.roster.swing.RosterFrameAction.actionPerformed(RosterFrameAction.java:55)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:761)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

pabender, 2017-01-03:
On 01/02/2017 08:52 PM, Randall Wood wrote:
> The RosterFrame also crashes during construction if the connection is
> "None":
>
> |2017-01-02 20:51:34,727 ptionhandler.UncaughtExceptionHandler ERROR -
> Uncaught Exception caught by [...]

Ok, that line is actually checking to see if the registered programmer 
manager supports the global programmer.  the None connection sets the 
programmers to null.

I've just made a change (and submitted a PR) to fix this one.

Paul

rhwood, 2018-03-07:
Likely fixed.

closed 2018-03-07 by rhwood