JMRI/JMRI #2807: Roster date parse Warning

silverailscolo, 2017-01-02:
On start up in 4.7.1ish I get the following warning in the Console for every Roster entry:
``roster.RosterEntry WARN  - Unable to parse May 3, 2016 10:09:46 AM as a date. [main]``
using German JMRI Locale and NL (Dutch) country system setting on macOS 10.12.2.

But the date is properly displayed in the Roster entry edit panel, so warning seems unnecessary:
[image]

rhwood, 2017-01-02:
Are the dates showing in the Roster table or are they blank there?

"May" is neither German nor Dutch for the fifth month of the year, which is why this is not parsable.

silverailscolo, 2017-01-02:
It's _not_ showing in the Roster table. We might learn from the Roster entry edit panel how to parse the date as stored to my Locale.

rhwood, 2017-01-02:
The roster entry edit panel isn't attempting to parse a date, it's only displaying a raw string. We have had multiple user requests since at least 2012 to use Dates in that table so they can be sorted, since the results of Date.toString() in other than the current locale is not sortable, so roster entries saved in 4.7.1 or newer store a parsable ISO 8601 date, but entries from earlier versions of JMRI do not.

The result is that once roster entries are saved in 4.7.1, they can be sorted in that table.

silverailscolo, 2017-01-02:
Had to try that, and it works fine. Should we add a hint to "Save the respective entry to fix this" to the console warning?

closed 2017-02-16 by silverailscolo