JMRI/JMRI #2808: SensorDefaultState start up warnning

silverailscolo, 2017-01-02:
On start up in 4.7.1ish I get the following warning in the Console for every Roster entry:
``2017-01-02 15:01:25,369 rix.AbstractSimulatorConnectionConfig WARN  - Loading found invalid value for option SensorDefaultState, found "Unknown", setting to "Unknown" [init prefs]``
using German JMRI Locale and NL (Dutch) country system setting on macOS 10.12.2.

Those two string look identical (and correct). Either hide warning or explain more.

bobjacobsen, 2017-01-02:
The problem here is that this bit of code in `jmri.jmrix.AbstractSimulatorConnectionConfig` is trying to set the current option value:

```
                JComboBox<String> opt = new JComboBox<String>(adapter.getOptionChoices(i));
                opt.setSelectedItem(adapter.getOptionState(i));
                // check that it worked
                if (!adapter.getOptionState(i).equals(opt.getSelectedItem())) {
                    // no, set 1st option choice
                    opt.setSelectedIndex(0);
                    adapter.setOptionState(i, (String) opt.getSelectedItem());
                    log.warn("Loading found invalid value for option {}, found \"{}\", setting to \"{}\"", i, adapter.getOptionState(i), opt.getSelectedItem());
                }
                options.put(i, new Option(adapter.getOptionDisplayName(i), opt, adapter.isOptionAdvanced(i)));
```

To do that, it's using the value from the XML file to set an option in a JComboBox.  If the XML file is in English (which in this case it seems to be), that won't match the _translated_ JComboBox in the interface.

This kind of thing (storing in one Locale, loading in another) has come up before, c.f. #2064, #1765. I don't think we have a general solution.

silverailscolo, 2017-01-03:
I suspect the warning is unclear about that because it displays the value of  ``adapter.getOptionState(i)`` after it was reset to the first item (default) 1 line earlier. Fix would be to put that line after this one or store the value in a variable. Will try.

silverailscolo, 2017-01-03:
Fix in #2815

closed 2017-01-21 by silverailscolo