JMRI/JMRI #2809: Sensor Groups won't work with userNames

KenC57, 2017-01-02:
It will silently convert them to system names and create new sensors to match. So if the rest of your panel is using the username you end up with the sensor group working with the system names it created and not your sensors.

bobjacobsen, 2017-01-21:
I've probably misunderstood the scenario, as I wasn't able to recreate it.  Could you give me an example, preferably with internal sensors?

What I tried (schematically):
- Create three sensors IS1, IS2, IS3
- give them user names
- Create a group by checking the three lines that have these sensors

And everything seemed fine.  So I've either missed the right way to create the group, or how to tell the group isn't working right.

KenC57, 2017-01-21:
I'm recalling that what I seemed to find in the code was it saved the system
name and not the user name. So if you did a move (or was it a rename) at
some point, I found they didn't match up any more. Granted USUALLY the
sensors in these groups are all internal, so doing moves (or renames) may
not be that typical. I was trying to review them since I found some not
working quite right. While it looks like the groups are implemented via
Logix, you have to read the xml to see what is really where, Logix won't let
you look at them. The sensor group does give a view but that's where I got
suspect of if it was keeping track by system or user name.

 

It would be a couple of days until I could tie back to that layout to
retrace what I was doing that made me come up with the issue.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2017-07-03:
Sensor groups create Conditionals (albeit a special SensorGroupConditional subtype) in a "System Logix" to do their work, and parse those Conditionas when you want to display or edit the result.

So this is really an Issue with "Conditionals not properly handling rename of references to sensor user names"

dsand47, 2017-07-04:
Sensor groups do work with user names, with certain restrictions.

- Sensor groups can be created when sensors have user names.
- If a user name is moved or added to a group sensor, the update references selection has to be **No**.
- If a user name is removed from a group sensor, the update references selection should be **Yes**.
- A group sensor can be renamed.

**Background**

A conditional action has a single value, DeviceName, which can contain either a system name **OR** a user name to refer to an object, in this case a sensor.

When a sensor group is created, the generated conditional actions all have sensor system names in DeviceName.

When the property change event occurs, the triggering sensor system name is compared to the DeviceName for each entry in the sensor group conditional.  All of the entires that do NOT match are set Inactive.  The sensor that triggered the actions is left Active.  If DeviceName contains a user name, all the sensors are inactive and the group is broken.

The only effective way that I have found to repair a sensor group is delete and re-create.

The root problem is not conditional reference change handling (which does work as expected), but the restriction in the sensor group logic to only use system names.

Dave Sand

bobjacobsen, 2017-07-04:
Thanks.  A couple questions:

> When a sensor group is created, the generated conditional actions all have sensor system names in DeviceName.

Is there a reason this can't be handled through the usual NamedBeanHandle method? That would (a) allow either kind of name to be specified and (b) all the various name-move operations to work, without restriction.

> When the property change event occurs, the triggering sensor system name is compared to the DeviceName for each entry in the sensor group conditional. 

Why not compare the sensor objects instead of names?  e.g. something like (in pidgeon form):
```
if (this.sensorHandle.getBean().equals(event.getSource())) ....
```

Then it doesn't matter which kind of name is being used.

KenC57, 2017-07-04:
I feel strongly that supporting userName is a requirement, not an option.
Use of the move option to change which systemName is connected to a userName
is a common tool a number of us have been recommending for years during
layout development. For the sake of the users, we need to keep consistency.
A number of times we've talked about ways that the systemName should be keep
hidden as much as possible in favor of the userNames.

 

That said, I can understand a limitation that when you do some of these
changes like moving things around, you need to save and restart to insure
all the listeners are on the same page so to speak.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dsand47, 2017-07-04:
Assuming that no one is working on this, I went ahead and created PR #3718 to fix this issue.

I think I managed to test all the permutations of name changes, although users can get quite creative.  

Dave Sand

bobjacobsen, 2018-01-23:
Closed by #3718

closed 2018-01-23 by bobjacobsen