JMRI/JMRI #2819: RosterTest method testReadWrite fails interimitently

rhwood, 2017-01-04:


rhwood, 2017-01-04:
I hit "enter" too soon, and it seems the initial comment was empty when GitHub sent the issue notification. You'll need to read it online, sorry about that.

The gist of the issue is that the roster index is sometimes written incorrectly. Samples of a bad write are in the issue online.

rhwood, 2017-01-08:
Update title to reflect real issue--this was tracked down to the roster index not being populated correctly and sometimes storing a single entry twice and failing to store the following entry.

#2820 attempts to address this by no longer using indexed iterators over the list of roster entries, however, that PR is exhibiting a ConcurrentModificationException.

bobjacobsen, 2017-01-28:
An example of ConcurrentModificationException is in #2945 where Linux graphical CI failed with
```
    [junit] Testcase: testAttributeList(jmri.jmrit.roster.RosterTest):	Caused an ERROR
    [junit] null
    [junit] java.util.ConcurrentModificationException
    [junit] 	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1380)
    [junit] 	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
    [junit] 	at jmri.jmrit.roster.Roster.getAllAttributeKeys(Roster.java:431)
    [junit] 	at jmri.jmrit.roster.RosterTest.testAttributeList(RosterTest.java:312)
```

rhwood, 2017-01-29:
I expect that #2957 will address this.

rhwood, 2017-02-04:
This appears to have been fixed in #2957.

closed 2017-02-04 by rhwood