JMRI/JMRI #2823: Cannot Start PanelPro unless all ancillary devices connected

Pugwash1, 2017-01-04:
In the last day a change has been made that will not allow JMRI to start unless all defined devices are connected. Normally we run with the Bachurus Speedo and the programming PR3 disconnected. (in fact packed away)

KenC57, 2017-01-04:
Steve,

 

That is sort of by design. If you have a system and sometimes you want one
set of connections and other times different, that is the purpose of
profiles. We've had too many issues where things go bad when a connection is
not found, including loss of panel file contents by users.

 

So make multiple profiles but point them to the same directory for things
like rosters, panel files, etc. They just start from different profiles to
keep the connection configuration correct for what you will have connected.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2017-01-05:
This is explicitly by design. See the discussion at https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/134146

Use multiple profiles, and select the correct profile for the hardware that is on when JMRI starts.

closed 2017-01-13 by rhwood