JMRI/JMRI #2828: Duplicate C/MRI connection tabs shown in tables

spassmo, 2017-01-05:
I'm seeing duplicate C/MRI connection tabs in the turnout, sensor, and light tables.   Also when adding a light C/MRI is listed twice in the system connection pull-down menu.

I tried renaming the C/MRI connection and the name changed on both tab labels.

[image]

[image]

bobjacobsen, 2017-01-28:
Thanks for all the detail, that's very helpful.  Could you add two more things?

1)  What shows on the Preferences -> Connections pane?  Are there two entries or just one? (A screen shot would help with details)

2) Go to Help -> Upload Debugging Info..., include #2828 (this Issue number) in the title, and upload your configuration with all the boxes checked (well, you can leave off the 'Send me a copy' if you want)

Thanks!

tgretton, 2017-01-28:
Bob

I won't be back to the club until Tuesday. I will send debugging info  and screenshots then.

There was only one entry on the Connections pane.

And my recollection was that the second C/MRI pane was empty.

Tom

spassmo, 2017-01-28:
That is correct,  there is only one entry in the connections pane and the second tab in the tables is empty.

I've uploaded the debugging info.

Steve

[image]

[image]

bobjacobsen, 2017-01-28:
Thanks for sending all that.  Really big setup!  (Have attached files here for the record)

I've got it starting up locally (though without the hardware), but I'm not seeing the same problem.  Does it happen consistently, every time you start up?

[Issue2828NapaNorthernUpload.zip](https://github.com/JMRI/JMRI/files/737193/Issue2828NapaNorthernUpload.zip)

bobjacobsen, 2017-01-28:
Aha!  Now I see it. (If anybody is following this:  Once you have it running, open the SensorTable) OK, working on it.

spassmo, 2017-01-29:
Thanks Bob for looking into this.  I'll follow this and be sure to help if I can.  I can log into the club computer remotely to test things if needed.

Steve

bobjacobsen, 2017-03-13:
Closed by #3233

closed 2017-03-13 by bobjacobsen