JMRI/JMRI #2829: W10 support ?

chopperaddict, 2017-01-05:
When I try to run decoder pro ( and panelPro as well) it runs somewhere in background according to task manager, but nothing is visible anywhere.

So I run the install test batch file, and not that it reports that my platform, which is W10 Home 64 bit is NOT SUPPORTED ??????

The things it is complaining about are os.name and os.arch which are reported by  the utility as "Windows 10" and "amd64" which is strange as my machine has an Intel 7 processor on board......

Can this be true ??  What is going on ??
[image]

[image]

Any help would really be appreciated...

rhwood, 2017-01-05:
We only Serialio to support the Digitrax MS100. The license we have to distribute Serialio limits us to supporting it only on the 32-bit editions of Windows XP.

Are you attempting to use a Digitrax MS100?

chopperaddict, 2017-01-07:
no, I use a Prodigy DCC system ?

*Ian Turner*
Forums handle -* ChopperAddict*
Specialist Helicopter builder and setup expert
Check my web site http://www.chopperaddict.co.uk

NB - Google, GCHQ, the US NSA and ALL other DATA COLLECTION OPERATIONS-
I have the absolute right to retain my personal privacy and YOU should
respect that....,
So kindly *CEASE FROM ARCHIVING **any of my data* ......  Thank you so much.

rhwood, 2017-01-10:
@chopperaddict Can you please run Panel Pro and after waiting a couple of minutes, copy the file C:\Users\olwenJMRI\log\session.log and attach it to this issue (you may have to do this at the Github website instead of replying to the email)? That might shed some light on what's happening.

rhwood, 2017-02-08:
Closing since no response.

closed 2017-02-08 by rhwood