JMRI/JMRI #2832: Withrottle Server window

joandecastro, 2017-01-05:
That is a suggestion to improve the window from Withrottle Window;

We use JMRI in our club, one of the fuctions it to suport smartphones, tablets and other wifi hardware.

We suggest to modify the window from withrottle server: Actually we can't resize the windows, so we only can watch 7 o eight withrottle/engine driver clients and their dcc adreess. In that case it apears a scrolling bar.
We suggest that the window can be resized, as all windos in JMRI so we can display more clients without using the scroll bar.

Thank you very much

mattharris, 2017-01-10:
@joandecastro PR #2857 should address this.

mattharris, 2017-01-12:
This change is now merged into `master`.

closed 2017-01-25 by rhwood