JMRI/JMRI #2833: LocoNet Monitor changed path before 4.7.1ish

KenC57, 2017-01-06:
I think it was somewhere in the 4.5.* era, but if your startup had the old class path, it works, but you also get this error in the console:

2017-01-05 21:25:12,379 startup.AbstractActionModel           ERROR - Connection  does not exist. Cannot be assigned to action jmri.jmrix.loconet.locomon.LocoMonPane$Default [main]

Deleting the entry and entering new works to remove the error message.

So should we be updating the class or capture the error and make a meaningful report to the user to update the startup preference. I think it has to do with the feature of telling which connection an action is using.
-ken c

rhwood, 2017-01-06:
Editing the entry to include the required information (which connection should be monitored) would also silence the error. The class or package name last changed in 2010, so that's not the cause of the issue; however the class did change to need a connection be associated with it, since its otherwise impossible to have multiple loconet connections and open a monitor on the correct connection. The problem is that the connection name is not wrapped in quotes, so its not clear that the connection "" does not exist.

KenC57, 2017-01-06:
Would it make sense that a startup item that expected a connection find a
default connection?? I get the idea (since the old entry still worked) that
it was really doing that, using the default connection.

 

But the correct cure is somehow letting the user know they need to update
their preference and select a connection for the action. Improved logging
message would be one step. Should it trigger a Edit->Preferences popup or
something visible to the user??

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2017-01-06:
The 80% solution is to ignore it, since most users have only a single connection. Where users have multiple connections, they are probably already setting things up correctly. If it's really a problem, the opened panel should handle notifying the user. We should change this to a warning instead of an error in any case.

KenC57, 2017-01-06:
I would include that updating the preference is the right thing to do. The
message isn't right just saying the value is blank. That's only the symptom.
The problem is the preference needs updating to make the selection

 

-ken c

rhwood, 2017-01-08:
Fixed in #2836 by displaying a warning instead of an error and noting what can be done about this in the logs. I am not forcing this to be addressed by the user, since the window will "just work" for a user with only a single connection.

closed 2017-01-08 by rhwood