JMRI/JMRI #2843: 4.7.1ish validate xml fails working panel: intermediate signal is true

KenC57, 2017-01-08:
Offending line:

          <value class="java.lang.Boolean">true</value>

Part of:
    <virtualsignalmast class="jmri.implementation.configurexml.VirtualSignalMastXml" userName="B7B6">
      <systemName>IF$vsm:NS-2008:CLS-3-hi($0018)</systemName>
      <userName>B7B6</userName>
      <properties>
        <property>
          <key>intermediateSignal</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
      <unlit allowed="no" />
    </virtualsignalmast>

Thrown error:
2017-01-08 17:33:48,670 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/kcameron/JMRI/CnyMod/CnyModMast46-mty.xml Exception: org.jdom2.input.JDOMParseException: Error on line 3942: cvc-complex-type.4: Attribute 'class' must appear on element 'key'.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 3942: cvc-complex-type.4: Attribute 'class' must appear on 
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)

I thought I entered this one already. I can't find it in the list.

bobjacobsen, 2017-01-09:
Where did the XML file come from?  E.g. what JMRI version wrote it?

KenC57, 2017-01-09:
That was created by 4.5.something. And it is also in a 4.6 file I wrote this
afternoon (on a Linux system running 4.6). All these tests are using my HEAD
system.

-ken c

rhwood, 2017-01-10:
@KenC57 For all of these validation issues you've filed, please attach the panel file that fails so we have a test case.

KenC57, 2017-01-11:
[CnyModMast46-mty.zip](https://github.com/JMRI/JMRI/files/698582/CnyModMast46-mty.zip)

rhwood, 2017-01-11:
Every panel I've encountered so far that is failing validation in 4.7.1 also fails to validate in 4.6.

KenC57, 2017-01-11:
The key difference is the panels load and seem to work just fine. The fact
that the schema was out of date (not current with code) means we must either
fix the schema or remove the test against it before loading a panel.

-ken c

PR #2950 (2017-01-28): ConfigXml Schema update and additional tests
- fixes layoutblocks 2.9.6 schema error and adds test file (closes #2848)

Add test files for already-working cases

- closes #2843 
- closes #2844 
- closes #2845  
- closes #2846  
- closes #2847 
- closes #2859 

As far as I know, this is the full set of schema issues and concerns before 4.7.1
closed 2017-01-28 by bobjacobsen