JMRI/JMRI #2844: 4.7.1ish xml validate fails on panel:

KenC57, 2017-01-09:
Note: if using the validate option instead of loading does not give a string easy to cut/paste and it does not make any log entry either. So you have to use the panel open to get something to cut/paste.

Log:
`2017-01-08 19:00:53,332 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/kcameron/JMRI/PanelProLocoNet/CentralTest.xml Exception: org.jdom2.input.JDOMParseException: Error on line 2317: cvc-enumeration-valid: Value 'center' is not facet-valid with respect to enumeration '[left, centre, right]'. It must be a value from the enumeration.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 2317: cvc-enumeration-valid: Value 'center' is not facet-valid with respect to enumeration '[left, centre, right]'. It must be a value from the enumeration.
`
Offending line:
`    <positionablelabel x="407" y="359" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" text="Central" size="12" style="1" red="51" green="51" blue="51" justification="center" class="jmri.jmrit.display.configurexml.PositionableLabelXml">
`

bobjacobsen, 2017-01-09:
Looks like a language issue:  `centre` vs `center`

But it's not a property file issue, because `jmri.jmrit.display.configurexml.PositionableLabelXml` actually writes `centre` at line 105, and has since at least February 2015:
```
40cca4bd7 java/src/jmri/jmrit/display/configurexml/PositionableLabelXml.java (Randall Wood    2015-02-21 13:12:15 +0000 104)             case 0x04:
40cca4bd7 java/src/jmri/jmrit/display/configurexml/PositionableLabelXml.java (Randall Wood    2015-02-21 13:12:15 +0000 105)                 just = "centre";
40cca4bd7 java/src/jmri/jmrit/display/configurexml/PositionableLabelXml.java (Randall Wood    2015-02-21 13:12:15 +0000 106)                 break;
```

Where did the panel file come from?

KenC57, 2017-01-09:
I was watching football and trying every panel file in one of my test
directories. It had LocoNet based files from many users. It was a file date
7/17/2015. I would have thought this was a case of something that crept past
the locale filter. I had thought we were making sure all the 'data' items
were of one locale and that the UI would translate those to whatever made
sense to the user.

-ken c

KenC57, 2017-01-11:
[CentralTest.zip](https://github.com/JMRI/JMRI/files/698585/CentralTest.zip)

rhwood, 2017-01-14:
Of note is that JMRI has always silently ignored the justification settings in this panel ("centre" is accepted by JMRI, but this file had "center") and set the justification to the left. #2876 allows JMRI to accept "center" or "centre", so this panel may display differently after that is merged.

PR #2950 (2017-01-28): ConfigXml Schema update and additional tests
- fixes layoutblocks 2.9.6 schema error and adds test file (closes #2848)

Add test files for already-working cases

- closes #2843 
- closes #2844 
- closes #2845  
- closes #2846  
- closes #2847 
- closes #2859 

As far as I know, this is the full set of schema issues and concerns before 4.7.1
closed 2017-01-28 by bobjacobsen