JMRI/JMRI #2845: 4.7.1ish panel xml validation fails:

KenC57, 2017-01-09:
Log entry:
`2017-01-08 19:04:48,168 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/kcameron/JMRI/PanelProLocoNet/CSX%20Plan%20One%20text.xml Exception: org.jdom2.input.JDOMParseException: Error on line 893: cvc-complex-type.2.4.a: Invalid content was found starting with element 'properties'. One of '{systemName, unlit, disabledAspects, outputs, bitStrings}' is expected.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 893: cvc-complex-type.2.4.a: Invalid content was found starting with element 'properties'. One of '{systemName, unlit, disabledAspects, outputs, bitStrings}' is expected.
`
Offending line:
`      <properties>
        <property>
          <key class="java.lang.String">intermediateSignal</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
`

Part of:
`    <signalmast class="jmri.implementation.configurexml.SignalHeadSignalMastXml" userName="S22AW">
      <systemName>IF$shsm:basic:one-searchlight(S22AW)</systemName>
      <userName>S22AW</userName>
      <comment>Glass Recycling West</comment>
      <properties>
        <property>
          <key class="java.lang.String">intermediateSignal</key>
          <value class="java.lang.Boolean">true</value>
        </property>
      </properties>
      <unlit allowed="no" />
    </signalmast>
`

bobjacobsen, 2017-01-09:
The schema (xml/schema//types/signalmasts-2-9-6.xsd) requires `<properties>` before `<comment>`, not after.  

The `jmri.implementation.configurexml.SignalHeadSignalMastXml` class uses `storeCommon` from `jmri.managers.configurexml.AbstractNamedBeanManagerConfigXML` to write those. It writes `<comment>` before `<properties>`, so the schema looks backwards.

silverailscolo, 2017-01-09:
(Partial) fix in PR #2810

KenC57, 2017-01-11:
[CSX Plan One text.zip](https://github.com/JMRI/JMRI/files/698587/CSX.Plan.One.text.zip)

PR #2950 (2017-01-28): ConfigXml Schema update and additional tests
- fixes layoutblocks 2.9.6 schema error and adds test file (closes #2848)

Add test files for already-working cases

- closes #2843 
- closes #2844 
- closes #2845  
- closes #2846  
- closes #2847 
- closes #2859 

As far as I know, this is the full set of schema issues and concerns before 4.7.1
closed 2017-01-28 by bobjacobsen