JMRI/JMRI #2858: xml validation error loading user preferences at startup

pabender, 2017-01-10:
This is from a profile I haven't started in several months:

org.jdom2.input.JDOMParseException: Error on line 62: cvc-complex-type.2.4.a: Invalid content was found starting with element 'properties'. One of '{width, height}' is expected.
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)
	at org.jdom2.input.SAXBuilder.build(SAXBuilder.java:1091)
	at jmri.jmrit.XmlFile.getRoot(XmlFile.java:171)
	at jmri.jmrit.XmlFile.rootFromURL(XmlFile.java:150)
	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:604)
	at jmri.configurexml.ConfigXmlManager.lambda$load$0(ConfigXmlManager.java:576)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: org.xml.sax.SAXParseException; lineNumber: 62; columnNumber: 19; cvc-complex-type.2.4.a: Invalid content was found starting with element 'properties'. One of '{width, height}' is expected.
	at org.apache.xerces.util.ErrorHandlerWrapper.createSAXParseException(Unknown Source)
	at org.apache.xerces.util.ErrorHandlerWrapper.error(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator$XSIErrorReporter.reportError(Unknown Source)
	at org.apache.xer

ces.impl.xs.XMLSchemaValidator.reportSchemaError(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator.handleStartElement(Unknown Source)
	at org.apache.xerces.impl.xs.XMLSchemaValidator.startElement(Unknown Source)
	at org.apache.xerces.xinclude.XIncludeHandler.startElement(Unknown Source)
	at org.apache.xerces.impl.XMLNSDocumentScannerImpl.scanStartElement(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl$FragmentContentDispatcher.dispatch(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XMLParser.parse(Unknown Source)
	at org.apache.xerces.parsers.AbstractSAXParser.parse(Unknown Source)
	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:217)
	... 20 more

The file identified as the cause is attached (though I had to give it a different extension... github doesn't like it as a .xml file)

[UserPrefsProfileConfig.txt](https://github.com/JMRI/JMRI/files/697195/UserPrefsProfileConfig.txt)
.

rhwood, 2017-01-11:
Did this file pass validation if manually checked in earlier versions? (I suspect not)

pabender, 2017-01-11:
Sent from my iPhone

> On Jan 11, 2017, at 2:11 PM, Randall Wood <notifications@github.com> wrote:
> Did this file pass validation if manually checked in earlier versions? (I suspect not)
> 
I don't know, and I am not going to check.  This file was written by a version prior to 4.4.

The date on the original file was in may.

Paul

bobjacobsen, 2017-01-28:
This validates against head of master.  I've added it as a test case via #2950

closed 2017-01-28 by bobjacobsen