JMRI/JMRI #2859: 4.7.1ish fails validation on "hasBackground" on a reporter icon

pabender, 2017-01-10:
[java] org.jdom2.input.JDOMParseException: Error on line 234: cvc-complex-type.3.2.2: Attribute 'hasBackground' is not allowed to appear in element 'reportericon'.
     [java] 	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:232)
     [java] 	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:253)
     [java] 	at org.jdom2.input.SAXBuilder.build(SAXBuilder.java:1091)
     [java] 	at jmri.jmrit.XmlFile.getRoot(XmlFile.java:171)
     [java] 	at jmri.jmrit.XmlFile.rootFromURL(XmlFile.java:150)
     [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:604)
     [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:568)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:224)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:167)
     [java] 	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:157)
     [java] 	at jmri.configurexml.LoadXmlConfigAction.loadFile(LoadXmlConfigAction.java:51)
     [java] 	at jmri.configurexml.LoadXmlUserAction.actionPerformed(LoadXmlUserAction.java:43)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
     [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java] 	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java] 	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
     [java] 	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
     [java] 	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
     [java] 	at java.awt.Component.processMouseEvent(Component.java:6533)
     [java] 	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
     [java] 	at java.awt.Component.processEvent(Component.java:6298)
     [java] 	at java.awt.Container.processEvent(Container.java:2236)
     [java] 	at java.awt.Component.dispatchEventImpl(Component.java:4889)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2294)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
     [java] 	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
     [java] 	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2280)
     [java] 	at java.awt.Window.dispatchEventImpl(Window.java:2746)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
     [java] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:731)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:729)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
     [java] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
     [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
     [java] Caused by: org.xml.sax.SAXParseException; lineNumber: 234; columnNumber: 300; cvc-complex-type.3.2.2: Attribute 'hasBackground' is not allowed to appear in element 'reportericon'.
     [java] 	at org.apache.xerces.util.ErrorHandlerWrapper.createSAXParseException(Unknown Source)
     [java] 	at org.apache.xerces.util.ErrorHandlerWrapper.error(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLErrorReporter.reportError(Unknown Source)
     [java] 	at org.apache.xerces.impl.xs.XMLSchemaValidator$XSIErrorReporter.reportError(Unknown Source)
     [java] 	at org.apache.xerces.impl.xs.XMLSchemaValidator.reportSchemaError(Unknown Source)
     [java] 	at org.apache.xerces.impl.xs.XMLSchemaValidator.processAttributes(Unknown Source)
     [java] 	at org.apache.xerces.impl.xs.XMLSchemaValidator.handleStartElement(Unknown Source)
     [java] 	at org.apache.xerces.impl.xs.XMLSchemaValidator.startElement(Unknown Source)
     [java] 	at org.apache.xerces.xinclude.XIncludeHandler.startElement(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLNSDocumentScannerImpl.scanStartElement(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl$FragmentContentDispatcher.dispatch(Unknown Source)
     [java] 	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
     [java] 	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
     [java] 	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
     [java] 	at org.apache.xerces.parsers.XMLParser.parse(Unknown Source)
     [java] 	at org.apache.xerces.parsers.AbstractSAXParser.parse(Unknown Source)
     [java] 	at org.jdom2.input.sax.SAXBuilderEngine.build(SAXBuilderEngine.java:217)
     [java] 	... 49 more

file attached (with modified extension)

[XBeeDemoLayoutPanel.txt](https://github.com/JMRI/JMRI/files/697220/XBeeDemoLayoutPanel.txt)

rhwood, 2017-01-11:
Every panel I've encountered so far that is failing validation in 4.7.1 also fails to validate in 4.6.

pabender, 2017-01-11:
Sent from my iPhone

> On Jan 11, 2017, at 2:15 PM, Randall Wood <notifications@github.com> wrote:
> 
> Every panel I've encountered so far that is failing validation in 4.7.1 also fails to validate in 4.6.
> 
That may be, but version 4.6 isn't enforcing the validation at load time and 4.7.1ish is.

The fact is that this file, and the ones Ken has posted were written by previous versions of JMRI and there is now a Mis-match between the schema and things JMRI has actually written. 

That is the issue that needs to be fixed to allow the validation, and subsequent loading, to function.

All of these files are test cases that must be able to be loaded before we can release 4.7.1.  

Paul

KenC57, 2017-01-11:
I've got likely 50 other panel files from users in my collection. Once we
think we've got this batch fixed I will resume testing of other panels and
submitting as needed to document other issues with the schema not matching
the program.

-ken c

rhwood, 2017-01-15:
@pabender Can this panel be included in the JMRI code as a test case?

petecressman, 2017-01-20:
fixed.

petecressman, 2017-01-20:
fixed - test case test.jmri.jmrit.display.configurexml.verify.reporter.xml

bobjacobsen, 2017-01-28:
The java/test/jmri/jmrit/display/configurexml/verify/reporter.xml file isn't the same as Paul's XBeeDemoLayoutPanel. No problem having both, so I'll add XBeeDemoLayoutPanel

PR #2950 (2017-01-28): ConfigXml Schema update and additional tests
- fixes layoutblocks 2.9.6 schema error and adds test file (closes #2848)

Add test files for already-working cases

- closes #2843 
- closes #2844 
- closes #2845  
- closes #2846  
- closes #2847 
- closes #2859 

As far as I know, this is the full set of schema issues and concerns before 4.7.1
closed 2017-01-28 by bobjacobsen