JMRI/JMRI #2879: LocoNet Se8c editor load fails on missing resource tag

KenC57, 2017-01-15:
Somewhere along the line since 4.6, it went missing. The tag 'READ FROM'

Here is the stack trace head:
`2017-01-15 08:09:11,457 swing.JmriNamedPaneAction             WARN  - could not load pane class: jmri.jmrix.loconet.se8.SE8Panel due to:java.util.MissingResourceException: Can't find resource for bundle java.util.PropertyResourceBundle, key READ FROM  [AWT-EventQueue-0]
java.util.MissingResourceException: Can't find resource for bundle java.util.PropertyResourceBundle, key READ FROM 
	at java.util.ResourceBundle.getObject(Unknown Source)
	at java.util.ResourceBundle.getString(Unknown Source)
	at jmri.jmrix.loconet.AbstractBoardProgPanel.provideAddressing(AbstractBoardProgPanel.java:181)
	at jmri.jmrix.loconet.se8.SE8Panel.<init>(SE8Panel.java:41)
	at jmri.jmrix.loconet.se8.SE8Panel.<init>(SE8Panel.java:35)
`
I've traced it to the removal of trailing spaces in the keys. I'm submitting a fix for the file since having a trailing space in the key sounds like a bad idea anyway.

devel-bobm, 2017-01-15:
The basic problem also breaks the LocoNet DS64, BDL16x, and PM4x configuration tools.

The <jmrix.loconet.AbstractBoardProgPanel> was intended to provide generic support for these four Digitrax devices, and was tweaked so that the JButtons would be customized to show the device name as part of the button label.  This was done to help differentiate the visual look of tools for different device types which otherwise look the same.

The current code has incorrectly implemented the bundle reference key generation:

<        readAllButton = new JToggleButton(java.util.ResourceBundle.getBundle("jmri/jmrit/symbolicprog/SymbolicProgBundle").getString("READ FROM ") + " " + type);
        writeAllButton = new JToggleButton(java.util.ResourceBundle.getBundle("jmri/jmrit/symbolicprog/SymbolicProgBundle").getString("WRITE TO ") + " " + type);>

perhaps instead needs to be

<        readAllButton = new JToggleButton(java.util.ResourceBundle.getBundle("jmri/jmrit/symbolicprog/SymbolicProgBundle").getString("READ FROM "type));
        writeAllButton = new JToggleButton(java.util.ResourceBundle.getBundle("jmri/jmrit/symbolicprog/SymbolicProgBundle").getString("WRITE TO " + type));>

And the following keys need to be in the bundle:

<READ FROM BDL16x
WRITE TO BDL16x
READ FROM DS64
WRITE TO DS64
READ FROM PM4x
WRITE TO PM4x
READ FROM SE8C
WRITE TO SE8C>

Those changes would restore the button label text to the previous behavior.

Perhaps this points out a problem where generating the "key" from concatenated strings, as done in <AbstractBoardProgPanel> breaks various bundle QC checks?

KenC57, 2017-01-15:
Paul,
The concatenation is correct. It is merging the locale string for 'Read From' with the board type so you would see 'Read From Se8c' or whichever board you are using. The board name (part name) doesn't change for different languages, just what the 'read from' should say. Your example would have it expecting different keys for each board type. Only issue I could see as written is if some languages wanted to do something like 'Se8c read it' as the proper text. I recall there is a proper way to use the resource strings but embed arguments into the translated string. So there is a way to deal with the order of the message in the button.
-ken c

devel-bobm, 2017-01-15:
@KenC57,

The previous functionality was designed so that the message could be internationalized.  It was coded many years ago, at a time when there were no JMRI guidelines for parameterizing localizable messages, or perhaps there was no such capability available.  I will rework the code to use the JMRI-documented  "parameterizable" concept.

devel-bobm, 2017-01-15:
The "Bundle" mechanisms previously-used in <jmri.jmrix.loconet.LocoNet AbstractBoardProgramming> were changed to reference jmri/jmrit/symbolicprog/SymbolicProgBundle.  That broke the loconet-specific functionality with robust internationalization capability.

silverailscolo, 2017-01-15:
When you use ``Bundle.getMessage("key", variable)`` the (translated) strings will be fetched from JMRI/java/src/jmri/jmrix/loconet/LocoNetBundle.properties and the extra parts will be inserted in any place where you use ``{0}``. This has proven to be a very robust method, with the great advantage of passing a search up the hierarchy so we can reuse existing keys across many packages. As for performance, I am told it is also faster. And it's a lot shorter. 
Looking at the se8c panel, there's still some i18n to be done, so there's much te be gained by moving from the symbolicprog props to this bundle. I could assist you if you wish.

devel-bobm, 2017-01-16:
@silverailscolo, I am already in-process of implementing this.  Per one of my previous comments, such features were not available or were not documented as JMRI recommended practices when this code was initially implemented.

KenC57, 2017-01-16:
I'll stay out of this then and wait for you all to figure it out.

-ken c

devel-bobm, 2017-01-16:
Should be fixed in development builds, beginning with build which is available at [http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1149/](url).

bobjacobsen, 2017-02-03:
Closing as fixed by #2881

closed 2017-02-03 by bobjacobsen