JMRI/JMRI #2903: File path problem

DanielBoudreau, 2017-01-21:
One of our users reported that he couldn't save his operation files.  I asked for the system console output and found a very strange behavior, the requested path name is being modified to a bogus one.  Here's the console:

2017-01-20 16:31:34,734 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2017-01-20 16:31:35,343 util.Log4JUtil INFO - This log is appended to file: C:\Documents and Settings\Administrator\JMRI\log\messages.log [main]
2017-01-20 16:31:35,359 util.Log4JUtil INFO - This log is stored in file: C:\Documents and Settings\Administrator\JMRI\log\session.log [main]
2017-01-20 16:31:35,609 apps.Apps INFO - PanelPro version 4.6-R81496dc starts under Java 1.8.0_111 on Windows 9X (unknown) x86 v5.1 at Fri Jan 20 16:31:35 CST 2017 [main]
2017-01-20 16:31:40,484 apps.Apps INFO - Starting with profile LCMRRC_Signal_and_Operations.3f3d0cc8 [main]
2017-01-20 16:31:42,359 node.NodeIdentity INFO - Using jmri-001D0989C31D-3f3d0cc8 as the JMRI Node identity [main]
2017-01-20 16:31:46,921 locobuffer.LocoBufferAdapter INFO - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true CD: false [main]
2017-01-20 16:31:54,640 configurexml.PerformFileModelXml INFO - Load file C:\Documents and Settings\Administrator\Desktop\LCMRRC JMRI FILES\CTC 1-18.xml [main]

2017-01-20 16:31:59,218 server.WebServer INFO - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=4.0} [WebServer]
2017-01-20 16:32:04,046 withrottle.UserInterface INFO - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleGUIServer]
2017-01-20 16:32:10,140 configurexml.TriggerRouteModelXml INFO - Setting route "LAYOUT RESET" at startup. [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path program: is C:\Program Files\JMRI\ [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path preference: is C:\Documents and Settings\Administrator\JMRI\ [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path profile: is C:\Documents and Settings\Administrator\JMRI\LCMRRC_Signal_and_Operations\ [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path settings: is C:\Documents and Settings\Administrator\JMRI\ [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path home: is C:\Documents and Settings\Administrator\ [main]
2017-01-20 16:32:11,593 util.FileUtilSupport INFO - File path scripts: is C:\Program Files\JMRI\jython\ [main]

2017-01-20 16:32:21,859 setup.BackupBase INFO - Saving copy of operations files to: C:\Documents and Settings\Administrator\JMRI\operations\autoBackups\2017_01_20_03 [AWT-EventQueue-0]
2017-01-20 16:32:38,500 util.FileUtilSupport INFO - Creating directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:32:38,515 util.FileUtilSupport ERROR - Failed to create directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:32:38,515 operations.OperationsXml ERROR - Exception while writing operation file, may not be complete: java.io.FileNotFoundException: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations\OperationsLocationRoster.xml (The filename, directory name, or volume label syntax is incorrect) [AWT-EventQueue-0]
2017-01-20 16:32:38,671 util.FileUtilSupport INFO - Creating directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:32:38,687 util.FileUtilSupport ERROR - Failed to create directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:32:38,687 operations.OperationsXml ERROR - Exception while writing operation file, may not be complete: java.io.FileNotFoundException: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations\OperationsCarRoster.xml (The filename, directory name, or volume label syntax is incorrect) [AWT-EventQueue-0]
2017-01-20 16:35:02,500 util.FileUtilSupport INFO - Creating directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:35:02,515 util.FileUtilSupport ERROR - Failed to create directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]
2017-01-20 16:35:02,515 operations.OperationsXml ERROR - Exception while writing operation file, may not be complete: java.io.FileNotFoundException: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations\Operations.xml (The filename, directory name, or volume label syntax is incorrect) [AWT-EventQueue-0]

This looks good:

2017-01-20 16:32:21,859 setup.BackupBase INFO - Saving copy of operations files to: C:\Documents and Settings\Administrator\JMRI\operations\autoBackups\2017_01_20_03

This doesn't look so good:

2017-01-20 16:32:38,500 util.FileUtilSupport INFO - Creating directory: C:\PROGRA~1\JMRI\.\C:\Documents and Settings\Administrator\JMRI\operations [AWT-EventQueue-0]

I see an "extra" C:\PROGRA~1\JMRI\. in the path name.  

Here's a copy of the code that's creating the directory:

   /**
     * Copies a complete set of Operations files from one directory to another
     * directory. Usually used to copy to or from a backup location. Creates the
     * destination directory if it does not exist.
     *
     * Only copies files that are included in the list of Operations files.
     * @param sourceDir From Directory
     * @param destDir To Directory
     *
     * @throws java.io.IOException Due to trouble reading or writing
     */
    public void copyBackupSet(File sourceDir, File destDir) throws IOException {
        log.debug("copying backup set from: {} to: {}", sourceDir, destDir);
        log.info("Saving copy of operations files to: {}", destDir);

        if (!sourceDir.exists()) // This throws an exception, as the dir should
        // exist.
        {
            throw new IOException("Backup Set source directory: " // NOI18N
                    + sourceDir.getAbsolutePath() + " does not exist"); // NOI18N
        }
        // See how many Operations files we have. If they are all there, carry
        // on, if there are none, just return, any other number MAY be an error,
        // so just log it.
        // We can't throw an exception, as this CAN be a valid state.
        // There is no way to tell if a missing file is an error or not the way
        // the files are created.

        int sourceCount = getSourceFileCount(sourceDir);

        if (sourceCount == 0) {
            log.debug("No source files found in {} so skipping copy.", sourceDir.getAbsolutePath()); // NOI18N
            return;
        }

        if (sourceCount != _backupSetFileNames.length) {
            log.warn("Only {} file(s) found in directory {}", sourceCount, sourceDir.getAbsolutePath());
            // throw new IOException("Only " + sourceCount
            // + " file(s) found in directory "
            // + sourceDir.getAbsolutePath());
        }

        // Ensure destination directory exists
        if (!destDir.exists()) {
            // Note that mkdirs does NOT throw an exception on error.
            // It will return false if the directory already exists.
            boolean result = destDir.mkdirs();

            if (!result) {
                // This needs to use a better Exception class.....
                throw new IOException(
                        destDir.getAbsolutePath() + " (Could not create all or part of the Backup Set path)"); // NOI18N
            }
        }

        // Just copy the specific Operations files, now that we know they are
        // all there.
        for (String name : _backupSetFileNames) {
            log.debug("copying file: {}", name);

            File src = new File(sourceDir, name);

            if (src.exists()) {
                File dst = new File(destDir, name);

                FileHelper.copy(src.getAbsolutePath(), dst.getAbsolutePath(), true);
            } else {
                log.debug("Source file: {} does not exist, and is not copied.", src.getAbsolutePath());
            }

        }

        // Throw a test exception, if we have one.
        if (testException != null) {
            testException.fillInStackTrace();
            throw testException;
        }
    }

Any idea why?

Dan

rhwood, 2017-01-21:
What are the user's preferences set to?

DanielBoudreau, 2017-01-21:
Not quite sure what you're asking for? Can that be determined from the console output?

rhwood, 2017-01-21:
Without the profile or operations preferences, what's going on cannot be determined. These preferences are not in the console output (for file paths, the resultant paths, not the actual preferences, are shown in the FileUtilSupport output in the console).

DanielBoudreau, 2017-01-26:
Randall,

Can you contact the user directly to get whatever you need?

Here's the link:

https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/135041

Dan