JMRI/JMRI #2910: When a Block Contents item is added to a Layout Editor panel it is invisible (SourceForge #513 )

geowar1, 2017-01-22:
(I'm bringing this over from SourceForge #513; originated by Dave Sand)
When a Block Contents item is added to a Layout Editor panel it is invisible.
The temporary work around is to add a value in the block table and then the item shows up. I then add a 1 one pixel border with a color so that it will stay visible.
--
These could be pre-populated with the name of the block… or any arbitrary text.
I'm willing to do the work… if a consensus can be reached as to what should be done.

KenC57, 2017-01-22:
Since we have the state of 'edit' available, we could add a number of
different ideas to make it show. Similar to the connection points, turnout
circles, track midpoints, etc..

 

I'd say a pixel boarder just outside the dimensions. That way it wouldn't
hide any existing details in the box.

-ken c

pabender, 2017-01-22:
Sent from my iPad

> On Jan 22, 2017, at 12:00 PM, George Warner <notifications@github.com> wrote:
> 
> (I'm bringing this over from SourceForge #513; originated by Dave Sand)
> When a Block Contents item is added to a Layout Editor panel it is invisible.
> The temporary work around is to add a value in the block table and then the item shows up. I then add a 1 one pixel border with a color so that it will stay visible.
> 
> These could be pre-populated with the name of the block… or any arbitrary text.
> I'm willing to do the work… if a consensus can be reached as to what should be done.
> 
My first reaction to this was we should leave it blank.  Unless the block contents have been defined, it really should its value as blank.

But maybe we need to sit back and think about what the block contents icon really is.  Functionally, it is somewhere between a reporter and a memory.  

Reporter icons will show <no report> if the reporter has no current report.  Reporter are actual hardware, so that makes sense.

It's been a while since I used one, but I believe a memory icon will also show blank if it has no current value.  So, like the block contents icon, it will be invisible by default.

I think that whatever we do, we want to be consistent across all three of these elements.  The border modification is probably the best approach, since it doesn't impact values of the underlying objects, which may be utilized in scripts or other logic.  It would be a good to allow the user to turn this feature off as well.

Paul

geowar1, 2017-01-22:
On Jan 22, 2017, at 1:09 PM, Paul Bender <notifications@github.com> wrote:
> My first reaction to this was we should leave it blank. Unless the block contents have been defined, it really should its value as blank.

The issue is that when it’s blank you can’t see it in the editor so you can’t move it or modify it unless you just happen to know where you created it.

> But maybe we need to sit back and think about what the block contents icon really is. Functionally, it is somewhere between a reporter and a memory.
>
> Reporter icons will show <no report> if the [...]

The “fix” might be to only add/draw the border when in edit mode?

-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

dsand47, 2017-01-22:
I totally forgot about this.

Here is some additional information:
[image]

- Row 1 -- Text boxes  :  The left one had no content when created.  It is not selectable.  The right one has content.  Deleting the content results in a warning text about **null** content, which to me is actually **empty** content.
- Row 2 -- Memory boxes :  The left one has a null value while the right one has an empty value (zero length).
- Row 3 -- Block boxes  :  The left one has a null value while the right one has an empty value.

The easiest way to find the invisible boxes is to do a drag selection on the area.
[image]

The empty boxes disappear completely outside of edit mode.
[image]

Once the empty boxes have been edited, they become visible regardless of mode.
[image]

I would recommend that the block boxes behave in the same manner as the memory boxes.  That way they are visible during editing regardless of state, but can be invisible or not when not in edit mode based on changes to defaults.

Creating an empty text box should result in a default content such as trying to empty an existing text box.

geowar1, 2017-01-26:
Looking at the sources: The text label is just a JLabel; there's no code in JMRI to draw it; it's handled by swing. A block box (BlockContentsIcon) is a subclass of a memory icon (MemoryIcon).
The memory & block boxes have routines (drawMemoryRects & drawBlockContentsRects) to draw an outline when in edit mode. These borders are drawn at the size of the PositionableLabel that they are both subclasses of. Their initial size is set to the getPreferredSize of the JComponent (swing super-class of PositionableLabel)... which appears to be the size of the text used to create the PositionalLabel… three (blank) spaces: "   " ([width=28,height=16]…)

geowar1, 2017-01-26:
Actually… I may have inadvertently solved this issue… By changing the memory and block text entry controls from JTextFields to JComboBoxes… and because the entered values are verified as valid… you can't create a memory or block label without some text in it. Blank or spaces won't work. Next development build play with it again and see if it works for you now.

dsand47, 2017-01-26:
A blank/empty (zero length string) value is, or should be, valid for both memory and block values.  It is also the normal state for unoccupied blocks.

An issue in the past is that these values are also **null** at times, most likely after being created.  This causes problems with Logix because it cannot test for null.

While beyond the scope of this particular request, I believe that new memory variables and block values should have a default of **empty** and that any process within JMRI that updates the block values use empty instead of null.  For example, block tracking when a block becomes unoccupied.

As of build #1182, using the new dropdown method, new block content fields are still invisible.  I will check when a new package build becomes available.

Dave Sand

bobjacobsen, 2017-01-26:
> An issue in the past is that these values are also null at times, most likely after being created. This causes problems with Logix because it cannot test for null.
>
> While beyond the scope of this [...]

‘null’ _is_ the Java way to express “empty”, in the sense of nothing contained.

A Memory value is an Object, so empty means “not containing an Object” which means null.

Many times (but not required to be all times; sometimes a Block contains e.g. a Throttle) Blocks contains Strings. Unfortunately, the String class’s concept makes this more complicated, because the ‘empty String' is still a String, hence different from null. This can confusing unless the language is kept _really_ clean.  

So there are two choices to fix the problem with Logix:

1) Give Logix (Conditionals) a way to test for null value

2) Change Blocks to have String values instead of Object values.

KenC57, 2017-01-26:
I think the trick in Logix to find null is the OR of being blank and being
not blank. Kind of like testing a sensor for not active OR not inactive, you
know it is undefined.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dsand47, 2017-01-26:
Here is a Logix test for a **null** memory variable:  `NOT Memory "name" >= value ""`

Thanks Ken for getting me started on looking for a solution.

Logix does not have direct access to block values.  This is done by linking a memory variable in the layout block definition.

A third option:  A new memory variable or block value has an initial value of blank/empty string.  After that, they can have a string or object, but never **null**.  One exception might be a Python script that intentionally uses null (_**None**_).

Dave Sand

rhwood, 2017-01-27:
> A new memory variable or block value has an initial value of blank/empty string. After that, they can have a string or object, but never null. One exception might be a Python script that intentionally uses null (None).

It must be possible to set Memories to null. An empty string is a value, and there are third party web applications that behave differently to an empty string from a null value.

KenC57, 2017-01-27:
Dave,

 

That shows we should add block values to Logix. I think around the last time
Logix was worked on, block values were only useful via the memories. But now
they are directly visible to the panels, they should be visible to the
Logix.

-ken c

dsand47, 2017-01-27:
Providing direct access to block values from Logix would be a good idea.  2 years ago, Dave Cochrun was having issues because Logix updated memory variables did not propagate the changes to the linked block value which breaks block tracking.  With direct access to the block values, the related memory variables are no longer required.

I don't have an issue with null values, but I don't think they should be a default value for memory variables and block values.

I have come up with an improved test for memory variables Logix:
`NOT Memory "name" > value ""`
By changing the test from >= to just >, either **null** or **empty** will test true.

Dave Sand

KenC57, 2017-01-27:
Dave,

 

One good reason for things like memories to be null is to show they never
had a value. Like a sensor that defaults to Unknown. During initialization
this is useful to know if it received a value yet. But handy tests like what
you showed that can treat the unknown (null) like an empty (blank) is good
for simplified coding.

-ken c

bobjacobsen, 2017-01-27:
> One good reason for things like memories to be null is to show they never had a value. Like a sensor that defaults to Unknown. During initialization this is useful to know if it received a value [...]

Agree with all the above.  Want to add on a bit.

For JMRI _itself_, not any particular use, it's important that Blocks, Memories and similar things be able to contain _Objects_, not just Strings.  If somebody wants to be able to put a Throttle into a Memory or Block, they should be able to.

The value of the Memory is a String or Throttle object.  If it's a String, that _String_ might in turn have a value of "" or "Foo" or something else.  

So if you're expecting a Block to contain a String, and it's got `null`, you might want to consider that to be "no problem, empty content" or you might want to consider that as "there's been a configuration error".  I don't care which, but I care very much that people be able to handle this either way if they want to.  That's yet another reason that Blocks, Memories, etc should be able to contain `'null`.

If there's a test that allows Logix/Conditionals to consider `null` object reference and empty string the same, that's great.  But it might still be useful to have a separate way of testing for `null`.  Of course, the Logix/Conditional code is dense and doesn't have good test coverage, so that might be very hard to do.

geowar1, 2017-02-02:
We've got a minor terminology issue: The value of a text label is the text assigned to that text label… however the value of a memory label is the value of the memory icon (icon?) assigned to that object (for example if the icon(?) assigned is "IMCURRENTTIME" it's value is the value of that variable (the current time)) and the value of a block label is the value of the object assigned to that block (for example the block icon(?) may be "Yard" and its value is the value of the object assigned to the "Yard" block (for example: "Train 424")).
So text labels have assigned text (that is the value of that label and is directly displayed); memory labels have assigned variables (that have a value that's displayed by the memory label) and block labels have assigned blocks (whose displayed value is the value of the block assigned to that label).
So we need to be specific about what we're talking about when we say "value"; a memory label's value is NOT the name of the memory location it represents: it's the value of that memory location and a block label's value is NOT the name of the block assigned to that label: it's the value of the object (if any) assigned to that block.

dsand47, 2017-02-02:
You are correct.

A text label is very different compared to a label to display memory variable or a block value contents.  For memory variables and block values, the label also acts as a data input source.

As discussed earlier, memory variables and block values can be null or contain an object (which can be a  string).  A text label is always a string.

Getting back to the original request, the block value label should be visible in edit mode just like the memory variable label regardless of the current content, if any.

Dave Sand

geowar1, 2017-02-06:
Would it be too confusing if the memory & block labels displayed their memory & block variables (not their values) (as text) while in edit mode?

dsand47, 2017-02-06:
I think it would be confusing for many people.  The tooltips already provide the name information.  It would make testing more difficult since you have to swap in and out of edit mode.

Dave Sand

pnbgit, 2017-10-12:
I wonder if the block value labels should default to having a single pixel border when created?  That way they would always be visible.  The user can remove it by editing the properties if they want.

I have also created issue #4247 where I have suggested we could have a "use these settings as default" when editing the properties for block values (could also apply to memory variables).

geowar1, 2019-11-04:
Has this issue been resolved? Can this be closed?(BTW: I was unable to create a "Block Contents" (block value label?); it fails silently… but I may be doing it wrong… can someone (knowledgeable) verify this?)

geowar1, 2019-11-04:
Nope; scratch that… they are being created… invisible… I think I can assume this has not been resolved…

closed 2021-04-14 by stale[bot]