JMRI/JMRI #2915: Duplicate decoder ID tets failing

bobjacobsen, 2017-01-22:
When run standalone, `jmri.jmrit.decoderdefn.DuplicateTest` reports a failure due to multiple TCS duplicates:
```
found duplicate for TCS DP2X decoder][DP2X
found duplicate for TCS DP2X decoder][DP2X
found duplicate for TCS DP2X decoder][DP2X-UK
found duplicate for TCS DP2X decoder][DP2X-UK
found duplicate for TCS Mx series decoders][M1
checked total of 2193
```

Note that this is _not_ included in our CI tests, as `DuplicateTest` is not included in `jmri.jmrit.decoderdefn.PackageTest`.  Once the duplicates are fixed, `DuplicateTest` should be added to `PackageTest` before this is closed.

rhwood, 2017-01-22:
This duplicates #2601

rhwood, 2017-01-22:
Note that per @dheap, this test is working correctly; the TCS decoder definitions are bad.

bobjacobsen, 2017-01-23:
The test is fine, agreed, but that means it's finding an issue, and it's not connected into the CI tests to keep more duplicates from being introduced.

bobjacobsen, 2017-01-23:
Ah, think I understand source of confusion: This is about getting the test run always during CI, _after_ the existing TCS issue (from #2601) is fixed.  Have changed title to make that clearer.

dheap, 2017-11-05:
#4372 resolves this issue and also re-enables the `DuplicateTest` in `pom.xml`.

Can this be closed now?

closed 2018-04-22 by dheap