JMRI/JMRI #2919: Panel Persistance Terminology and Duplicate Handling

bobjacobsen, 2017-01-23:
After [Yet Another Long JmriUsers Thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/134905) about how JMRI's panel persistence model works, I'd like to propose a renovation.  That [thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/134905) provides a lot of background.  This Issue is meant to be a place to accumulate discussion for a while, then create a proposal that we'll implement and live with for a while, hopefully a long time.  

First draft of the proposal:

1. Use "Load" and "Store" terminology throughout, at least for the English version.  (No more "Save" or "Open")

2. Add a warning for the specific case of loading the exact same file twice: “You’ve already loaded this file into JMRI, are you sure you want to do that again? This can create duplicates of panels.” with choices “Continue” (default) and “Cancel”. (Notice how “load” makes this make sense from the start: It’s much harder to say in terms of “open”)

3. Change the message on the dialog box that appears when you click the “close” button on a panel (e.g. red circle in upper left corner on a Mac, or similar on Windows and Linux)
   - It now says "This panel has been changed. Please remember to save it to disk. (Select 'Save Panels...' in the Panels menu or the File menu of any panel)” with buttons “Don’t show again”, “Delete Panel”, “Hide Panel”.
   - Instead, have it say “Do you want to hide this so it can later be reopened from the Windows menu, or do you want to remove it completely? (When you store your panel into a file, this will become permanent)” with choices “Hide” (default), “Remove” and “Cancel”, and a checkbox that says “remember this choice and don’t show again”. 
   - I’d actually prefer to leave off the “(When you store ..)” comment in the interest of brevity, and perhaps add a “ more info” button, but I understand that people don’t tend to use those.
![reminder_and_jmri_projects_jmri_ _java_ _java_-classpath__usr_local_cellar_ant_1_9_2_libexec_lib_ant-launcher_jar_-dant_home__usr_local_cellar_ant_1_9_2_libexec_-dant_library_dir__usr_local_cellar_ant_1_9_2_libexec_lib_org_apache_tools_ant_](https://cloud.githubusercontent.com/assets/2774117/22190316/666a247e-e0d7-11e6-8dc7-e6b647fa89aa.png)
(Needs a little work on sizing and formatting)

4. Documentation changes to match.

Note that this Issue is not about handling errors when e.g. a connection fails so that necessary pieces are missing when a file is loaded (see how that terminology works?). That should get a separate conversation that's independent of this.

rhwood, 2017-01-23:
I think this is merely papering over the issue that its not clear where configuration is stored. We should not allow a file to opened or loaded multiple times. While it may be convenient for some, I think its so far caused more trouble than benefits. (Maybe we could allow opening a single file once though API, but not through any menu action so "power users" who do take advantage of opening a single file multiple times can do so using a script.)

One thing that would be immensely clarifying would be not allowing hidden panels. If someone doesn't want to see a panel, they can minimize it, using the minimize button available in the OS (we can remember that state and re-open the panel minimized if need be). That would remove the entire overloading the "close" button problem, so the "close" button leads to a confirm the panel should be deleted or abort closing it.

Any point where we remind a user to save a panel, we should instead have a message "This panel has been changed. Save it now?" with Save/Cancel buttons.

dsand47, 2017-01-23:
I have a couple of ideas for consideration.

- Move the panel delete process from the panel close button to a **Delete >** menu item under Panels with the appropriate message text.  The close button retains the **Hide** function.
- Separate the panel management from the file management.

**Panels**
New  >
Show >
Delete >
---sep--- 
Load...
Store...
History
---sep---

- Move the file management menu items from the **Panels** menu to the **File** menu, although this might be too disruptive.

***Other thoughts***

Some people see a panel as a document.  When they load a configuration, they see their panel which looks like a drawing.  They don't automatically see all of the table data and internal data structures that makes the panel functional.

The general expectation is that when a document is closed, it can be re-opened.  The **Hide** action of the close button and the **Show >** menu item provide this capability within the JMRI context.

For true documents, deletion requires deleting the file which is independent of window buttons.  By moving the delete process to a menu item, deleting a panel now requires an intentional action. 

Ideally, the word **Panel** should no longer be associated with data persistence and file management.  The only word I can come up with so far is **Configuration** which might be too generic.  It could also be confused with **Preferences** which are stored in profiles.  

Dave Sand

KenC57, 2017-01-23:
I too have wondered about the hide/delete metaphor and that somehow
isolating the delete to a different point in the system might be safer.

 

Now and the name panel, I frequently use the word 'pane' for any single
window. So some of what Dave has said brings to mind if the whole idea of
panel, and the different thoughts of what is a panel, maybe we face the
large task to using some name changes to help convey the new idea of how
things would work. I really don't like the thought, I'd hope education would
be easier, but given how some of the discussions with users have gone over
the years, I'm not sure. Either way is littered with problems or baggage
over the years.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

dsand47, 2017-01-23:
For me, a **panel** is a specific kind of window that provides a _**canvas**_ on which a graphical representation of a layout and its controls is created using one of the three editors.

The above definition causes some people to think in traditional document terms.  Hence the issues with the close button, along with the concepts around load/open and store/save.

The Help documents, along with the Yahoo forum, makes a significant amount of educational material available.  Some of the issues I see are encouraging people to read the help material and finding the information in the Yahoo forum.  It is much easier to just ask the question again rather than searching for previous answers.

What is unknown is how many people are having issues vs. how many people might be impacted by changing things.

Dave Sand

closed 2021-04-14 by stale[bot]