JMRI/JMRI #2931: Can't delet a Preference under the edit window

bo4127, 2017-01-25:
Running Windows 8,1 with Decoder Pro 4.6 and JAVA 1.8.  When I open Decoder Pro and set the 1st connection under edit-preferences window, ever thing is ok.  However,  when I try to enter a 2nd connection preference and try to save everything, when it restarts only the 1st connection is there.  The 2nd is gone.  In fact I can't delete the 1st connection.  It always starts with the 1st and I can't ever add and save any more.

bo4127

KenC57, 2017-01-25:
Bo4127,

 

Which types of connections are you trying to load?? There is a problem with
creating or editing any CMRI type connection on 4.6.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bo4127, 2017-01-25:
I have the 1st one connected to a NCE POWERCAB.  The 2nd one is Digitrax PR3 to a DCS50.  Every time I save, it keeps the 1st one and not the 2nd even thou I delete the 1st before the save. When it reboots, the 1st one is back and the  second one is gone.  This is on a DELL laptop with Windows 8.1.  I have a older DELL running XP and Decoder Pro 3.4. with no problem.bo4127  On 01/25/17, Ken Cameroncom> wrote: Bo4127, Which types of connections are you trying to load?? There is a problem withcreating or editing any CMRI type connection on 4.6. -Ken Cameron, Member JMRI Dev Teamwww.jmri.orgwww.fingerlakeslivesteamers.orgwww.cnymod.comwww.syracusemodelrr.org

KenC57, 2017-01-25:
Bo4127,

 

Please open the system console before trying to edit the preferences. I
suspect something is happening and the log will give us clues. After you did
the edit and save, BUT don't restart, cut/paste the console and send us what
it said.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bo4127, 2017-01-26:
Attached are 2 log files.  The 1st is after starting Decoder Pro and then adding a 2nd preference but no save.  The 2nd log file is after I deleted the 1st preference but no save. I did not have it connected to a port  but that should make no difference.  It does the same thing if I have it connected to a port bo4127 On 01/25/17, Ken Cameroncom> wrote: Bo4127, Please open the system console before trying to edit the preferences. Isuspect something is happening and the log will give us clues. After you didthe edit and save, BUT don't restart, cut/paste the console and send us whatit said. -Ken Cameron, Member JMRI Dev Teamwww.jmri.orgwww.fingerlakeslivesteamers.orgwww.cnymod.comwww.syracusemodelrr.org

KenC57, 2017-01-26:
Bo,

 

The email strips attachments. So that's why I mentioned cut/paste the logs
to (one or more) emails. The other option is to use the 'Upload debugging
info' in the help menu. That will also send them via a web method that gets
to us as a package.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bo4127, 2017-01-26:
This is the console after the first open with no edit of the preference.log4j:ERROR setFile(null,false) call failed.java.io.FileNotFoundException: C:\Users\William\JMRI\log\session.log (Access is denied) at java.io.FileOutputStream.open0(Native Method) at java.io.FileOutputStream.open(Unknown Source) at java.io.FileOutputStream.<init>(Unknown Source) at java.io.FileOutputStream.<init>(Unknown Source) at org.apache.log4j.FileAppender.setFile(FileAppender.java:294) at org.apache.log4j.FileAppender.activateOptions(FileAppender.java:165) at org.apache.log4j.config.PropertySetter.activate(PropertySetter.java:307) at org.apache.log4j.config.PropertySetter.setProperties(PropertySetter.java:172) at org.apache.log4j.config.PropertySetter.setProperties(PropertySetter.java:104) at org.apache.log4j.PropertyConfigurator.parseAppender(PropertyConfigurator.java:842) at org.apache.log4j.PropertyConfigurator.parseCategory(PropertyConfigurator.java:768) at org.apache.log4j.PropertyConfigurator.configureRootCategory(PropertyConfigurator.java:648) at org.apache.log4j.PropertyConfigurator.doConfigure(PropertyConfigurator.java:514) at org.apache.log4j.PropertyConfigurator.configure(PropertyConfigurator.java:440) at jmri.util.Log4JUtil.configureLogging(Log4JUtil.java:175) at jmri.util.Log4JUtil.initLog4J(Log4JUtil.java:105) at jmri.util.Log4JUtil.initLogging(Log4JUtil.java:69) at jmri.util.Log4JUtil.initLogging(Log4JUtil.java:57) at apps.AppsBase.preInit(AppsBase.java:64) at apps.gui3.Apps3.preInit(Apps3.java:63) at apps.gui3.dp3.DecoderPro3.preInit(DecoderPro3.java:114) at apps.gui3.dp3.DecoderPro3.main(DecoderPro3.java:108)2017-01-26 10:48:26,711 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2017-01-26 10:48:26,726 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\William\JMRI\log\messages.log [main]2017-01-26 10:48:26,726 util.Log4JUtil                        INFO  - This log is stored in file: null [main]2017-01-26 10:48:26,742 apps.AppsBase                         INFO  - DecoderPro version 4.6-R81496dc starts under Java 1.8.0_121 on Windows 8.1 x86 v6.3 at Thu Jan 26 10:48:26 EST 2017 [main]2017-01-26 10:48:36,259 gui3.Apps3                            INFO  - Profiles not configurable. Using fallback per-application configuration. Error: C:\Users\William\JMRI\DecoderProConfig3.properties (Access is denied) [main]2017-01-26 10:48:37,759 node.NodeIdentity                     ERROR - Unable to store node identities: C:\Users\William\JMRI\nodeIdentity.xml (Access is denied) [main]2017-01-26 10:48:37,759 node.NodeIdentity                     INFO  - Using jmri-90489A99B21D-3f6b6e92 as the JMRI Node identity [main]2017-01-26 10:48:39,119 usbdriver.UsbDriverAdapter            ERROR - Serial port COM4 not found [main]2017-01-26 10:48:39,119 configurexml.ErrorHandler             ERROR - COM4 not found while opening connection [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ThrottleManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.ThrottleManager instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.CommandStation instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.PowerManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.PowerManager instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ConsistManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.ConsistManager instance [main]2017-01-26 10:48:44,510 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.managers.ManagerDefaultSelector: System connection NCE provides a null manager for interface jmri.ConsistManager [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\William\JMRI\ [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\William\JMRI\PanelPro\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\William\JMRI\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path home: is C:\Users\William\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2017-01-26 10:48:51,480 throttle.PowerManagerButton           INFO  - No power manager instance found, panel not active [main]2017-01-26 10:48:52,308 gui3.TabbedPreferencesAction          INFO  - We are already waiting f [... cut ...] ces\JmriSRCPServerPreferences.xml to new format in C:\Users\William\JMRI\PanelPro\profile. [Thread-4]   On 01/25/17, Ken Cameroncom> wrote: Bo, The email strips attachments. So that's why I mentioned cut/paste the logsto (one or more) emails. The other option is to use the 'Upload debugginginfo' in the help menu. That will also send them via a web method that getsto us as a package. -Ken Cameron, Member JMRI Dev Teamwww.jmri.orgwww.fingerlakeslivesteamers.orgwww.cnymod.comwww.syracusemodelrr.org

bo4127, 2017-01-26:
This is the system console after adding the digitrax preference and deleting the NCE preference with no save.log4j:ERROR setFile(null,false) call failed.java.io.FileNotFoundException: C:\Users\William\JMRI\log\session.log (Access is denied) at java.io.FileOutputStream.open0(Native Method) at java.io.FileOutputStream.open(Unknown Source) at java.io.FileOutputStream.<init>(Unknown Source) at java.io.FileOutputStream.<init>(Unknown Source) at org.apache.log4j.FileAppender.setFile(FileAppender.java:294) at org.apache.log4j.FileAppender.activateOptions(FileAppender.java:165) at org.apache.log4j.config.PropertySetter.activate(PropertySetter.java:307) at org.apache.log4j.config.PropertySetter.setProperties(PropertySetter.java:172) at org.apache.log4j.config.PropertySetter.setProperties(PropertySetter.java:104) at org.apache.log4j.PropertyConfigurator.parseAppender(PropertyConfigurator.java:842) at org.apache.log4j.PropertyConfigurator.parseCategory(PropertyConfigurator.java:768) at org.apache.log4j.PropertyConfigurator.configureRootCategory(PropertyConfigurator.java:648) at org.apache.log4j.PropertyConfigurator.doConfigure(PropertyConfigurator.java:514) at org.apache.log4j.PropertyConfigurator.configure(PropertyConfigurator.java:440) at jmri.util.Log4JUtil.configureLogging(Log4JUtil.java:175) at jmri.util.Log4JUtil.initLog4J(Log4JUtil.java:105) at jmri.util.Log4JUtil.initLogging(Log4JUtil.java:69) at jmri.util.Log4JUtil.initLogging(Log4JUtil.java:57) at apps.AppsBase.preInit(AppsBase.java:64) at apps.gui3.Apps3.preInit(Apps3.java:63) at apps.gui3.dp3.DecoderPro3.preInit(DecoderPro3.java:114) at apps.gui3.dp3.DecoderPro3.main(DecoderPro3.java:108)2017-01-26 10:48:26,711 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2017-01-26 10:48:26,726 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\William\JMRI\log\messages.log [main]2017-01-26 10:48:26,726 util.Log4JUtil                        INFO  - This log is stored in file: null [main]2017-01-26 10:48:26,742 apps.AppsBase                         INFO  - DecoderPro version 4.6-R81496dc starts under Java 1.8.0_121 on Windows 8.1 x86 v6.3 at Thu Jan 26 10:48:26 EST 2017 [main]2017-01-26 10:48:36,259 gui3.Apps3                            INFO  - Profiles not configurable. Using fallback per-application configuration. Error: C:\Users\William\JMRI\DecoderProConfig3.properties (Access is denied) [main]2017-01-26 10:48:37,759 node.NodeIdentity                     ERROR - Unable to store node identities: C:\Users\William\JMRI\nodeIdentity.xml (Access is denied) [main]2017-01-26 10:48:37,759 node.NodeIdentity                     INFO  - Using jmri-90489A99B21D-3f6b6e92 as the JMRI Node identity [main]2017-01-26 10:48:39,119 usbdriver.UsbDriverAdapter            ERROR - Serial port COM4 not found [main]2017-01-26 10:48:39,119 configurexml.ErrorHandler             ERROR - COM4 not found while opening connection [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ThrottleManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.ThrottleManager instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.CommandStation instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.PowerManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.PowerManager instance [main]2017-01-26 10:48:44,510 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ConsistManager to null value [main]2017-01-26 10:48:44,510 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for NCE (class jmri.jmrix.nce.NceSystemConnectionMemo) provides a null interface jmri.ConsistManager instance [main]2017-01-26 10:48:44,510 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.managers.ManagerDefaultSelector: System connection NCE provides a null manager for interface jmri.ConsistManager [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\William\JMRI\ [main]2017-01-26 10:48:50,386 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\William\JMRI\PanelPro\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\William\JMRI\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path home: is C:\Users\William\ [main]2017-01-26 10:48:50,402 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2017-01-26 10:48:51,480 throttle.PowerManagerButton           INFO  - No power manager instance found, panel not active [main]2017-01-26 10:48:52,308 gui3.TabbedPreferencesAction      [... cut ...] 10:54:32,787 jmrix.AbstractSerialPortController    ERROR - No usable ports returned [AWT-EventQueue-0]  On 01/25/17, Ken Cameroncom> wrote: Bo, The email strips attachments. So that's why I mentioned cut/paste the logsto (one or more) emails. The other option is to use the 'Upload debugginginfo' in the help menu. That will also send them via a web method that getsto us as a package. -Ken Cameron, Member JMRI Dev Teamwww.jmri.orgwww.fingerlakeslivesteamers.orgwww.cnymod.comwww.syracusemodelrr.org

bobjacobsen, 2017-01-28:
One problem is that `(Access is denied)` error in the logs: Some directory that JMRI is going to write in has been protected somehow.  I don't know enough about Windows to be certain, but it looks like the permissions are not allowing JMRI to write to `C:\Users\William\JMRI\`, which is something that it really does need to do.

bo4127, 2017-01-29:
I saw the same thing.  However, I set the permission level for all files in the JMRI directory to all users to full permission for all levels for the files in JMRI directory and the directory. It still gives the access denied for writing to the file.  It did however, allow writing to other files that were denied access before I changed the access to full for anybody.bo4127Bill On 01/28/17, Bob Jacobsencom> wrote: One problem is that (Access is denied) error in the logs: Some directory that JMRI is going to write in has been protected somehow.  I don't know enough about Windows to be certain, but it looks like the permissions are not allowing JMRI to write to C:\Users\William\JMRI\, which is something that it really does need to do.

bo4127, 2017-01-29:
I should also comment that this only happens when I am trying to add/delete a preference under PanelPro and not DecoderPro.  I can have only 1 preference in PanelsPro and if I want to add another or delete the only one after restart , the same one is restored.  I can add/delete multi preferences in DecoderPro.

bo4127, 2017-01-30:
The access denied message happens with both the DecoderPro and PanelPro opens.Ihave set the access to all the files to all users. Is this a problem with Java. The last time the session.log file was opened was on 1/7/16 at10:4:22.  I don't remember what happened at that time because I never tried to add/delete a connection preference in either DecoderPro or PanelPro.  Attached is a messages.log file which contain the open of PanelPro and I try to add another preferenceand then a save.  The I opened PanelPro  again and the new preference is gone. look like it goes in a loop for over 3 seconds trying to save the new preference. 

[messages.docx](https://github.com/JMRI/JMRI/files/737977/messages.docx)

closed 2017-01-30 by bo4127