JMRI/JMRI #2933: Double slip needs 2 control points and each point needs the (also throw option)

lwaldie, 2017-01-25:
Double slips require two control points so as to not inadvertently line other turnouts that are connected to one or more of there legs acting as a crossover , when you cycle through the single control point. crossovers have switch motors on each point wired to the same DCC address so they both line at the same time and if the first movement is on an adjacent crossover that you don't want lined it could cause a DERAIL, also with no (also throw option) on the points of a slip when you use the control point of the slip acting as a crossover the other turnout even if it throws wont show in the layout editor graphics if you don't have a switch wired to verify movement, witch I don't. that then affects other trains running in dispatcher mode.

geowar1, 2017-01-25:
Slime me…

bobjacobsen, 2017-01-28:
Closed by #2934

closed 2017-01-28 by bobjacobsen

geowar1, 2017-02-04:
Just an FYI: this issue should be fixed in the latest 4.7.1ish build at: <http://jmri.tagadab.com/jenkins/job/Development/job/Packages/>.
Please beat on it and let me know if you have any issues. (I'm a JMRI/java noob so I'm not 100% confident that all my changes didn't break something… ;-)