JMRI/JMRI #2942: NamedBean must not be null error

JasonJanzen, 2017-01-27:
I am using HEAD (from about 1/22/17). When removing a signal mast at a end bumper (changing to no selection in the drop down box) the following occurs:  (does not occur in my 4.6 download)

2017-01-27 10:24:34,270 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException: bean must be nonnull
	at java.util.Objects.requireNonNull(Objects.java:228)
	at jmri.NamedBeanHandleManager.getNamedBeanHandle(NamedBeanHandleManager.java:52)
	at jmri.jmrit.display.layoutEditor.PositionablePoint.setEastBoundSignalMast(PositionablePoint.java:452)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.removeSignalMastAssignment(LayoutEditorTools.java:8223)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.removeSignalMastFromPanel(LayoutEditorTools.java:8287)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.setSignalMastsAtBoundaryDonePressed(LayoutEditorTools.java:8360)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.access$3100(LayoutEditorTools.java:62)

JasonJanzen, 2017-01-28:
I have made corrections to the setEastBoundSignalMast method of PositionablePoint to accept  and assign a null mast and thus skipping the call to getNamedBeanHandle.  Preliminary testing appears good, but have not created pull request yet as unsure this is the proper correction.  (same change to west bound)
 public void setEastBoundSignalMast(String signalMast) {
        SignalMast mast = null;
        if (signalMast != null && !signalMast.equals("")) {
            mast = InstanceManager.getDefault(jmri.SignalMastManager.class).getSignalMast(signalMast);
            if (mast == null) {
                log.error("Unable to find Signal Mast " + signalMast);
                return;
            }
// added following else clasue
        } else {
            eastBoundSignalMastNamed = null;
            return;
        }

bobjacobsen, 2017-01-28:
(It's easier to look at things like this if you commit and push to your repository, then send a link; that way we can be sure to be understanding the nesting of the code.  Also, using [triple-back-quote formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/) will leave the spacing intact)

I think this looks like a good solution. 

The problem is that lines like this:
```
eastBoundSignalMastNamed = InstanceManager.getDefault(jmri.NamedBeanHandleManager.class).getNamedBeanHandle(signalMast, mast);
```
are asking the `NamedBeanHandleManager` to do something intrinsically inconsistent:  "We want to match up the signalMast name with the null reference from now on".  

As I understand your code, it handles the "name and null reference" case by forcing eastBoundSignalMastNamed to null (which the rest of the code seems to properly interpret) and skipping the set.

Are there west, north and south cases that need the same treatment?

bobjacobsen, 2017-02-03:
@JasonJanzen Is there a fix we can merge for this? Thanks!

JasonJanzen, 2017-02-04:
There is something here I do not understand.  I had pushed my fix to my repository but comparison with JMRI/JMRI showed no differences.  Looking at the code I see the change I was trying to make already in JMRI/JMRI repository.  It seems like to me that some one else picked it up and created a pull.  Just a guess.

bobjacobsen, 2017-02-04:
Do you think you committed and pushed from your "master" branch?  That shows no change since 22 days ago:
[image]
Or was it perhaps some other branch and I'm looking at the wrong one?

JasonJanzen, 2017-02-04:
That is the correct branch.

JasonJanzen, 2017-02-04:
I am beginning to understand.  I have gotten confused between signal heads and signal masts when looking at the code.  I will try this change again.

JasonJanzen, 2017-02-04:
Ok pull request created.  Don't know where I went wrong before. Sorry.

rhwood, 2017-02-07:
Fixed by #2996.

closed 2017-02-07 by rhwood