JMRI/JMRI #2963: Possible issue with defaults on Loksound definitions

pabender, 2017-01-30:
On a non-JMRI site, I've seen a couple of reports of performing a "Write All Sheet" to a new LokSound Micro Select and an Intermountain SD40-2 with sound (which uses what I believe is a Loksound Micro Select decoder in a locomotive specific format) rendering a decoder inoperable.  

The reports I've seen say the decoder will not respond to the reset commands.

The users were able to get the decoders back to life by reprogramming them with an ESU Lok Programmer.

I'm trying to get more information about which specific decoder definition was used to cause this.  We may have the wrong defaults in the definition and/or the wrong reset process encoded in the file.

In the meantime, I've recommended to the users that if they have more decoders to program, they perform a read all sheets first.

dheap, 2017-01-30:
There are as many sets of "factory defaults" for LokSound decoders as there are sound projects. We are talking about maybe around a thousand sets of defaults for around 1,000 CVs. Neither JMRI nor ESU's own LokProgrammer software can handle that directly.

When you load a sound project file to an ESU decoder, the LokProgrammer software rewrites the "factory defaults", based on those in the sound project file.

There are currently two ways around the myriad-of-defaults problem:
1. Read Decoder Data (LokProgrammer) or Read All Sheets (JMRI) before reprogramming.
2. Load the relevant Project File from the ESU website (LokProgrammer). For JMRI, you can use the Tools->Export CV LIst command in LokProgrammer to produce a text file that JMRI can import into a roster entry using the File->Import->LokProgrammer CV List File command.

The problems with option 2 above are:
1. You need access to Windows to install LokProgrammer. (You don't need the LokProgrammer hardware).
2. You need to know exactly which sound project is in your decoder.
3. The sound needs to be an ESU provided public one, not a proprietary third party one.

The sound project name is stored in the decoder as a text string of around 30 (not near computer to check) characters. When I eventually get time to tidy up and merge the code I wrote in 2014, JMRI will be able to read this string and the user can search for that exact string on the ESU website. Once we get this far, there are further possible options I have under consideration. Life keeps getting in the way of my plans...

dheap, 2017-01-30:
> The reports I've seen say the decoder will not respond to the reset commands.

I haven't had any problems using the JMRI Reset command to reset ESU decoders and I've used it a lot.

dheap, 2017-01-30:
Clarification:
- The Full Decoder Reset "CV8 to 8"  from the Resets Menu works fine to get the decoder back to as-supplied state, unless the retailer or user who loaded the sound project unchecked the "Overwrite Defaults" button in LokProgrammer. That has been known to happen. 
- The "Reset to Defaults" button on the Roster Entry pane followed by a Write All Sheets will cause problems and there are no easy solutions to that.

pabender, 2017-01-30:
Thanks @dheap .

I've pointed the users who are having trouble to this issue.  Hopefully they can comment about their experience here.

Paul

arwhyte, 2019-01-31:
Earlier this evening I experienced this issue while using DecoderPro to program a Bowser Executive Line U25B equipped with a LokSound Select OEM decoder with sound.  From the basic tab I clicked the "Reset to defaults" button and then clicked "Write all sheets".    Although the locomotive would respond to the throttle and move forwards and backwards, all sound was lost.  

The only way I was able to restore the "lost" sound was by following @dheap's suggestion to open the Decoder Pro Programmer and then perform a full reset by selecting _Reset > Factory Reset > Reset all CVs_ from the DecoderPro menu bar.

bobjacobsen, 2019-01-31:
This _might_ be due to one or more wrong default values in the decoder definition. 

Could you do a “read all sheets” from your working locomotive, export that roster entry, and attach the file here please?  You might have to add a .txt suffix to the name.

dheap, 2019-01-31:
> @arwhyte Earlier this evening I experienced this issue while using DecoderPro to program a Bowser Executive Line U25B equipped with a LokSound Select OEM decoder with sound. From the basic tab I clicked the "Reset to defaults" button and then clicked "Write all sheets". Although the locomotive would respond to the throttle and move forwards and backwards, all sound was lost.

@arwhyte Unfortunately what you did is a _very bad thing_ to do and will certainly wipe out sound and functions. As I outline [above](https://github.com/JMRI/JMRI/issues/2963#issuecomment-276109755) (and have done so many times on both the jmriusers and loksound groups) setting correct defaults in ESU definitions is (for all practical purposes) impossible. Even ESU's own LokProgrammer doesn't attempt to do it - it can only do it if you can determine, find and load the original sound project file. (There is a way of transferring this information into JMRI, but haven't time to explain just now)

The best thing to do is the full factory reset, "Read Full Sheet" on the CVs Pane (Reading Changes if needed to pick up read errors, save and use that as your base for programming.

bobjacobsen, 2019-01-31:
Would it help to add an option to the decoder definition allowing it to turn off the Teset To Defaults button?

dheap, 2019-01-31:
> Would it help to add an option to the decoder definition allowing it to turn off the Reset To Defaults button?

There may be an even better option. Something like an option that would make the Reset to Defaults button instead bring up the hardware Resets menu then (if the user proceeds) follow by a warning dialog to Read All CVs (maybe giving the option to initiate such)...

dheap, 2021-03-16:
Reset to Defaults button can now be disabled in decoder definition by an optional "allowResetDefaults" attribute in the decoder-config file. Setting it to "no" disables the "Reset to defaults" button and provides an explanatory tooltip. #8944

closed 2021-03-16 by dheap