JMRI/JMRI #2964: WiThrottle Server window enhancement?

devel-bobm, 2017-01-30:
A WiThrottle Server user requested the ability to "drop" a locomotive from a WiThrottle device via manipulation of the JMRI "WiThrottle" window.  This request was given to me by an attendee of the 2017 Amhurst MRR club show while visiting the JMRI help desk.

Presumably, a new button would be added to the row containing "Device Name" and "Address", which, when activated, would dis-associate the device from the given loco address.  For cases where a given device is associated with multiple throttles, a separate button would be provided for each logical throttle's address.  Presumably when such a feature is activated, the loco would first be set to speed 0, and then the logical connection with the Wi-device would be terminated.

rhwood, 2017-01-31:
Would it be better to make that a context (right click) menu element so it's more difficult to accidentally click?

devel-bobm, 2017-01-31:
I do not have strong opinions on specific GUI decisions.

I do not use WiThrottle server, so I do not plan to take on the task of coding changes for this issue.

mstevetodd, 2017-02-01:
Bob, what is the use case for this? I'm not following when this feature would be used. 
WiThrottle can be set to automatically disconnect (and drop all locos) if no communication occurs in a specified time.

devel-bobm, 2017-02-01:
I do not understand the specific use-case.  While this doesn't help much, I know that the person who made the original request is working with an NCE system.

If I can reach him, I will inquire and report back.

xaodxkobyhot, 2017-02-01:
Hi devel

My name is Frank

I am the one that was inquiring about a way to drop a loco that someone is using via WiThrottle.  When I use WiThorttle from laptop I can see what locos they are running, but I can't find a way to stop them from running that loco. 

I wanted to see if there is a way to add a button or a scroll down menu so I can drop that loco from the program via laptop.

rhwood, 2017-02-01:
@xaodxkobyhot Is the use case that a WiThrottle is controlling a locomotive already controlled by another throttle, that an unknown (and unexpected) WiThrottle has joined JMRI, or both?

xaodxkobyhot, 2017-02-01:
@rhwood   I can have different WiThrottle running the same locos. But there is no way to kill one of the locos through my computer on the program

greenwizard88, 2017-02-20:
Why not just use another throttle to steal that loco from the device?

I don't believe that the WiThrottle protocol has the ability to send loco stolen type of commands down to the throttle, so we would probably want to update the protocol. That's an entirely different can of worms as it could potentially break any older apps.

Do we want the ability to steal from WiThrottle server? What about adding engines to a specific throttle from the WiThrottle window? I think this is a fundamental difference between using WiThrottle as a "throttle server" that doesn't really have much control, and WiThrottle as a command station for smartphone throttles (essentially).

BHoffman351, 2017-02-20:
> On Feb 20, 2017, at 1:05 PM, greenwizard88 <notifications@github.com> wrote:
> I don't believe that the WiThrottle protocol has the ability to send loco stolen type of commands down to the throttle, so we would probably want to update the protocol.
> 
Well, sort of it does.  But for what purpose?
> Do we want the ability to steal from WiThrottle server?
> 
Would this mean that JMRI is stealing from itself?  The WiThrottle server is part of JMRI and the throttle technically is there. The apps just control that throttle.
> What about adding engines to a specific throttle from the WiThrottle window? I think this is a fundamental difference between using WiThrottle as a "throttle server" that doesn't really have much control, and WiThrottle as a command station for smartphone throttles (essentially).
> 
Is the desired effect to remove the ability to acquire and release locos from the WiThrottle and Engine Driver apps?

> But there is no way to kill one of the locos through my computer on the program

But I still do not understand what the use of this is.  Are we looking for a way to disconnect a troublesome operator?  I would shy away from changes to the protocol until an understanding of the real issue becomes clear.

Regards,
Brett

closed 2021-04-14 by stale[bot]