JMRI/JMRI #2967: unneeded Resource Names in CmriActionListBundle_cs_CZ.properties ?

devel-bobm, 2017-01-31:
`jmri.jmrix.cmri.CmriActionListBundle_cs_CZ.properties` contains several Resource Names and associated strings which relate to systems other than CMRI.  Should these be removed?  Or moved to properties files in directories for the appropriate system?

I'm thinking that the following keys are mis-placed:
`jmri.jmrix.loconet.locormi.LnMessageServerAction
jmri.jmrix.easydcc.easydccmon.EasyDccMonAction
jmri.jmrix.lenz.swing.mon.XNetMonAction
jmri.jmrix.loconet.locomon.LocoMonPane$Default
jmri.jmrix.loconet.slotmon.SlotMonPane$Default
jmri.jmrix.nce.ncemon.NceMonAction
jmri.jmrix.zimo.zimomon.Mx1MonAction`

pabender, 2017-01-31:
my gut reaction is yes, those are miss-placed.

bobjacobsen, 2017-03-12:
I don't see that file at all. Has it been removed?  Can we close this?

closed 2017-07-03 by bobjacobsen