JMRI/JMRI #2974: Correction to Acknowledgements pane to remove broken link

Walter71342JMRI, 2017-02-01:
[Acknowledgements.zip](https://github.com/JMRI/JMRI/files/743634/Acknowledgements.zip)

mattharris, 2017-02-02:
@Walter71342JMRI This is starting from an earlier version of the `Acknowledgements.shtml` file than current, hence is not immediately easy to determine what's changed.

I _think_ that it's the EngineDriver link that's been altered - could you confirm?

Walter71342JMRI, 2017-02-02:
Yes  that the link that doesn't work.

How should I do this so that it shows what I changed?

Walt

Walter71342JMRI, 2017-02-02:
I am bow testing at the latest build from
http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

Walt

Walter71342JMRI, 2017-02-02:
> I am bow testing at the latest build from http://jmri.tagadab.com/
> jenkins/job/Development/job/Packages/
>
> Walt
>
> On Thu, Feb 2, 2017 at 7:58 AM, Walter Thompson <

LinkCecker 8.2 is showing 36 494 errors of the 737  it is checking. I am
sure there is more files to check. But the scan is stopping well looking at
http://jmri.org/help/en/webindex.shtml, line 1724 well tying to link to
http://jmri.og/help/en/hmtl/doc/technical/XmView.shtml

And I can not find this page to see what wrong.

bobjacobsen, 2017-02-05:
Thanks for trying to clean those links up.

The webindex.shtml and webtoc.shtml files are automatically generated from the XML files that control the JavaHelp index and table of contents, respectively.  Any edits to them get overwritten the next time a release is done, or perhaps sooner.  

So if there are problems in them, that's probably also a problem in the in-application index or TOC, which is fixed using the JHelpDev tool.  See our [documentation page on this](http://jmri.sourceforge.net/help/en/html/doc/Technical/Help.shtml).

Walter71342JMRI, 2017-02-07:
I don't think I really want to have access to the code. It looks like I would need that to be able to fix some of these. 
Walt

Walter71342JMRI, 2017-02-07:
And that is way I do not feel I should have access to the code!

bobjacobsen, 2017-02-12:
> And that is way I do not feel I should have access to the code!

This is why we use version control systems like GitHub.  You _can't_ mess up the common code.  

You get a copy on your own machine and work with it.  You can mess that up, but if you do, you just throw it away and start over.  Once you have a change that you think is right, you create a "pull request" (PR) on GitHub.  That's _not_ a change to the code on GitHub.  Nothing can be broken by creating the PR.  The system tests it, somebody reviews it, and if there are no problems, merges it in. If there are problems, you hear about them and can fix them. 

This system is completely safe, lets you fix things, and saves work for everybody.

http://jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml

closed 2017-03-13 by bobjacobsen