JMRI/JMRI #2975: Corrections to CreateDecoderAdvanced.shtml to remove broken link

Walter71342JMRI, 2017-02-01:
[CreateDecoderAdvanced.zip](https://github.com/JMRI/JMRI/files/743659/CreateDecoderAdvanced.zip)

dheap, 2017-02-01:
@Walter71342JMRI This patch is based on an outdated version of `CreateDecoderAdvanced.shtml`.

I updated this file quite a bit last week. You are probably using the version in the help directory of a previous JMRI release.

I've found the link issue in the latest version and modified my local checkout. I have a dinner appointment in a few minutes so I'll submit a PR tomorrow.

Meanwhile, would you like to check out both the [website](http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml) and a recent [development build](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) and let me know if there is something else that needs fixing. If so, let me know and I'll add it to the PR.

Dave

Walter71342JMRI, 2017-02-01:
Dave H:
Did not think to test with the latest buiilds. Will do that from now on,
(If I remember to).

I am running "LinkChecker" from Sourcefourg and try to make sense of what I
find as issue there. https://sourceforge.net/projects/linkchecker/

Walt

Walter71342JMRI, 2017-02-01:
I am running the linkchecker against both the Online and installed Help.

T*his may be creating issues for the other languages translations!!!!!!*!!!

Walt

dheap, 2017-02-02:
Closed via #2984.

closed 2017-02-02 by dheap