JMRI/JMRI #2976: PR #2939 causes stack overflow upon loading panel file

petecressman, 2017-02-01:
load loops at:
	at jmri.jmrit.logix.OBlock.hashCode(OBlock.java:186)
	at java.util.Objects.hashCode(Unknown Source)
	at java.util.HashMap$Node.hashCode(Unknown Source)
	at java.util.AbstractMap.hashCode(Unknown Source)
	at java.util.AbstractList.hashCode(Unknown Source)
	at java.util.Objects.hashCode(Unknown Source)
	at java.util.HashMap$Node.hashCode(Unknown Source)
	at java.util.AbstractMap.hashCode(Unknown Source)
	at java.util.Objects.hashCode(Unknown Source)
	at jmri.jmrit.logix.OBlock.hashCode(OBlock.java:186)
	at java.util.Objects.hashCode(Unknown Source)
	at java.util.HashMap$Node.hashCode(Unknown Source)
	at java.util.AbstractMap.hashCode(Unknown Source)
	at java.util.AbstractList.hashCode(Unknown Source)
	at java.util.Objects.hashCode(Unknown Source)
	at java.util.HashMap$Node.hashCode(Unknown Source)
	at java.util.AbstractMap.hashCode(Unknown Source)
	at java.util.Objects.hashCode(Unknown Source)
	at jmri.jmrit.logix.OBlock.hashCode(OBlock.java:186)
etc.....

petecressman, 2017-02-01:
I suggest removing line 186

bobjacobsen, 2017-02-01:
Could you attach the file please? Might have to append .txt to the name. Thanks.

bobjacobsen, 2017-02-01:
@rhwood: working from phone, can't test, but it looks like #2939 now has OBlock violating the hashcode() contract that equals() objects always have identical hash values.

rhwood, 2017-02-01:
I think this is now fixed, but can you please attach the panel that triggered this so we can include in the unit tests?

rhwood, 2017-02-01:
It's not that the contract was violated, it's that a circular reference for the hashcode could be created since an oblock contains a map of paths, and a path can have a reference to an oblock, so we went in a circle trying to generate the hashcode for an oblock->map->path->oblock->...

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

petecressman, 2017-02-02:
Several of my files had the same load error.  This one is a test file that does not require any connection type

rhwood, 2017-02-02:
@petecressman you'll have to visit GitHub to attach the file (follow the link at the bottom of the email) and upload it through the website (or email it directly to me and I'll add it). Sorry about that.

rhwood, 2017-02-07:
Fixed by #2979 and #2997.

closed 2017-02-07 by rhwood