JMRI/JMRI #2982: Layout zoom should be persistent across loads

geowar1, 2017-02-01:
First thing I do after loading a panel (or launching PanelPro) is kick the zoom up to x1.5. I'd like for this to ether persist across launches and/or per-panel load.

Comments? (I'm willing to implement this if there is interest.)

KenC57, 2017-02-04:
I've seen at least 3 users who would like it to be persistent, like other default settings, frame size, position, etc... but I'd like to see it also have a variable entry instead of the dozen fixed sizes. A few times one isn't quite big enough but the next is too big.
-ken c

pnbgit, 2017-02-09:
Yes please - I'd like this too, and I know one other user who wants it as well.  if setZoom could be made public whilst you are at it, so that the zoom can be set from a Jython script, that would also be very useful.

KenC57, 2017-02-09:
If somebody is working on the zoom code, I'd also ask about making it take a
value to zoom instead of just a few selected values. I know of some users
who would use about 133% if available. Being able to set the value (while
keeping the typical ones for the keystroke method) would be nice.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

geowar1, 2017-02-09:
On Feb 9, 2017, at 5:49 AM, Ken Cameron <notifications@github.com> wrote:
> If somebody is working on the zoom code, I'd also ask about making it take a
> value to zoom instead of just a few selected values. I know of some users
> who would use about 133% if available. Being able to set the value (while
> keeping the typical ones for the keystroke method) would be nice.

I’ll take on the zoom issues…

Please read and comment <https://github.com/JMRI/JMRI/issues/2982 <https://github.com/JMRI/JMRI/issues/2982>> if your zoom enhancement request isn’t already listed there. TIA.

-- 
Enjoy,
George Warner, (408)249-7533
eMail/iChat: <geowar1@mac.com>

pnbgit, 2017-02-09:
One other little tweak would be nice - at the moment the ctrl+ and ctrl- shortcut keys only work on the numeric keypad.  Laptops and compact keyboards (which we are using on one large layout) don't have a numeric keypad, so it would be really nice if ctrl+ and ctrl- on the main keyboard worked as well - which also would make it consistent with browser behaviours.  

Thanks,

Pete

geowar1, 2017-02-09:
Just looking at the sources I notice that the min/max zoom values are 0.25 and 6.0… any requests to change these?

wolfsonga, 2017-02-09:
With the ever increasing use of tablets for controlling turnouts, it would help if when launching a panel and setting it for web display, if the zoom could be controlled to adjust it to fit the target tablet. This becomes very important when a mixture of tablet sizes are used and when logging on are identified and then have the image going to that tablet adjusted for it's size. It should at least be adjustable from a jython script so that a button can be provided on the panel for adjustment after launch.

rhwood, 2017-02-09:
The zoom for a tablet should be configurable on the tablet, since its very easy to have a diverse set of tablets with diverse resolutions and zoom requirements. I’d think that problem is a separate problem from the zoom within the JMRI window (I’d also think that if the zoom within the JMRI window changes the display on the tablet, that’s a bug).

KenC57, 2017-02-09:
Since that works out to ¼ or 6/1 size, I think thats about the range of
what can be seen well. Id be really surprised to see an example needing
more of less of that range. Another way to think of it is to multiple the x
or y dimensions of the panels, you start to get real wild numbers quickly at
the extremes.

 

Over the 13 years Ive been around here, I dont recall hearing any ask for
outside that range (at least since it went to .25 to 6x). I recall around
2003-2005 or so was a change about either adding zoom or to adjust the range
and no change I recall since then.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

wolfsonga, 2017-02-09:
Randall,
  Sorry, I should have explained my purpose a little better. We want to
lock the tablets in kiosk mode to prevent accidental changes to the view if
someone swipes across or uses two fingers. There for the screen received
will be locked in position and cannot be resized. Being able to resize with
script commands would allow me to adjust for each tablet as it logs into
the JMRI web server and then it would stay at the setting throughout its
operation. That is my intent,
=gerry=

rhwood, 2017-02-09:
Gerry:

Your needs really need to be a separate request, as you need to be able to intercept and modify generated output from the web server while in transit, which is radically different than changing how JMRI stores local display settings.

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

rhwood, 2017-02-09:
Can the zoom state be stored per-computer?

I can easily imagine that someone might want a panel zoomed in 200% on a 27" monitor attached to a fixed workstation, but zoomed out to 0.75% on a Surface Pro.

(Under the hoods, we have mechanisms in place to do this already, but they are incomplete for panels.)

geowar1, 2017-02-10:
pnbgit : quick note: Currently the menu items can only have one shortcut assigned to them… (my Java Fu isn't yet good enough to figure out how to fix that… yet ;-) by changing the virtual code from "ADD" to "EQUALS" (and adding the shift modifier) I was able to change to the keyboard plus ('+') instead of the keypad plus.
So I changed it in the (US) LayoutEditorBundle.properties; if non-US users want it fixed they'll have to modify their (language specific) LayoutEditorBundle_XX.properties files:

# use ADD & SUBTRACT for keypad, EQUALS & MINUS for keyboard
zoomOutAccelerator = MINUS
zoomInAccelerator = EQUALS

KenC57, 2017-02-10:
Randall,

 

Are you implying the zoom isn't part of the panel but just the profile
specific user defaults??

 

At least that's what I would have expected. Persistence is part of the
preferences which in turn is part of the profile. So that's not impacting
the panel, at least that's my hope of this stuff.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

geowar1, 2017-02-10:
KenC57: Currently the zoom isn't saved/restored anywhere… it always comes up x1.0. This issue is to address that (and fix the keyboard shortcuts; and make the setZoom() API public (so it can be scripted) and make the zoom up/down increments finer grained.)
Integration tests are running as we speak… ;-)

wolfsonga, 2017-02-10:
Okay, you are right. Sorry I didn't catch that. Will do it when I have time
to write it up. Thanks.
=gerry=

KenC57, 2017-02-10:
I'm thinking that moving items like scrollbar control, zoom, and a number of other features (even edit) into the user preferences, so it could be per-profile, would be a great idea for how to implement the kiosk mode.

This would allow a club computer to have a default profile that gets launched on startup that has everything cooked in and editing off. So the smarter folk would be able to change the profile selected and restart and then do the editing and work on the panels. It's my morning coffee thought of how this may have started with zoom, but then picked up things like menu bars, scrollbars, etc... and what might be the best way for them all. Yes right now those become features of the panel, but it is looking more like the user preferences is the best place for this sort of thing. All these items were created before we had any concept like profiles and only very limited user preferences. So this looks like the right direction we should go (how far, how fast, that's another story).
-ken c

geowar1, 2017-02-11:
Hey KenC57, can you write that up as a new issue? Please be specific about which preferences you think should be migrated.
I'll take assignment of it if you do.
I'd like to close this one…

KenC57, 2017-02-11:
Ok George, I've created issue #3039 which summarized where I think we are
now. I also mentioned the same features should be considered for the other
editors too. But I know PE/CPE are one body of code and LE is the other,
which you have figured out how to start with this stuff. Maybe somebody else
will be able to follow the ideas but inside the PE/CPE code.

-ken c

geowar1, 2017-02-14:
Just submitted pull request # https://github.com/JMRI/JMRI/pull/3059

geowar1, 2017-02-15:
Looks like this is DONE! ;-)

closed 2017-02-15 by geowar1