JMRI/JMRI #2988: 4.7.1ish-201702022119-jenkins-R14c6576 crashing in Logix adding Signalhead

Walter71342JMRI, 2017-02-03:
Adding a SigalHead as a State Variable in Logixs
Doesn't fail with RI.4.7.1ish-201702011943-jenkins-R799049d.exe

System console:
2017-02-02 22:44:15,216 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-02-02 22:44:15,223 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Walt\JMRI\log\messages.log [main]
2017-02-02 22:44:15,223 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Walt\JMRI\log\session.log [main]
2017-02-02 22:44:15,227 apps.Apps                             INFO  - PanelPro version 4.7.1ish-201702022119-jenkins-R14c6576 starts under Java 1.8.0_121 on Windows 10 x86 v10.0 at Thu Feb 02 22:44:15 CST 2017 [main]
2017-02-02 22:44:17,370 apps.Apps                             INFO  - Starting with profile HOn3.3ec971a3 [main]
2017-02-02 22:44:17,488 node.NodeIdentity                     INFO  - Using jmri-64006A824CD2-3ec971a3 as the JMRI Node identity [main]
2017-02-02 22:44:17,761 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2017-02-02 22:44:18,590 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2017-02-02 22:44:18,590 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Walt\Dropbox\jmri\HOn3\ [main]
2017-02-02 22:44:18,590 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Walt\JMRI\HOn3\ [main]
2017-02-02 22:44:18,591 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Walt\JMRI\ [main]
2017-02-02 22:44:18,591 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Walt\ [main]
2017-02-02 22:44:18,591 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Walt\Dropbox\jmri\py scripts\ [main]
2017-02-02 22:44:18,694 util.prefs                            WARN  - Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5. [init prefs]
2017-02-02 22:44:34,754 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.0 [AWT-EventQueue-0]
2017-02-02 22:44:34,756 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_121 [AWT-EventQueue-0]
2017-02-02 22:44:51,043 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
2017-02-02 22:45:08,830 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.util.MissingResourceException: Resource not found
	at jmri.Bundle.retry(Bundle.java:191)
	at jmri.Bundle.handleGetMessage(Bundle.java:150)
	at jmri.Bundle.handleGetMessage(Bundle.java:130)
	at jmri.Bundle.getMessage(Bundle.java:67)
	at jmri.ConditionalVariable.getStateString(ConditionalVariable.java:795)
	at jmri.jmrit.beantable.LogixTableAction.variableTypeChanged(LogixTableAction.java:3685)
	at jmri.jmrit.beantable.LogixTableAction.access$400(LogixTableAction.java:105)
	at jmri.jmrit.beantable.LogixTableAction$38.itemStateChanged(LogixTableAction.java:2426)
	at javax.swing.JComboBox.fireItemStateChanged(Unknown Source)
	at javax.swing.JComboBox.selectedItemChanged(Unknown Source)
	at javax.swing.JComboBox.contentsChanged(Unknown Source)
	at javax.swing.AbstractListModel.fireContentsChanged(Unknown Source)
	at javax.swing.DefaultComboBoxModel.setSelectedItem(Unknown Source)
	at javax.swing.JComboBox.setSelectedItem(Unknown Source)
	at javax.swing.JComboBox.setSelectedIndex(Unknown Source)
	at javax.swing.plaf.basic.BasicComboPopup$Handler.mouseReleased(Unknown Source)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at javax.swing.plaf.basic.BasicComboPopup$1.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

Walter71342JMRI, 2017-02-03:
Opps. Had to go back to JMRI.4.6-R81496dc.exe before it would work OK.

bobjacobsen, 2017-02-03:
This is because this this line:
```
  return Bundle.getMessage("SignalHeadTypeHeld"); // NOI18N
```
is requesting a property string that's not defined.  It should be defined in `java/src/jmri/NamedBeanBundle.properties`

Here's the ["git blame" output](https://github.com/JMRI/JMRI/blame/master/java/src/jmri/ConditionalVariable.java):
[image]

Looks like line 795 came from [this commit](https://github.com/JMRI/JMRI/commit/74049bafcaa6b5f2941360192c65113c42dc8e34)

@silverailscolo: Perhaps the string was supposed to be `SignalHeadStateHeld` instead of `SignalHeadTypeHeld`?

bobjacobsen, 2017-02-03:
I'm going to go ahead and use the existing `SignalHeadStateHeld` string. If that's not the best choice, we can fix that later.

silverailscolo, 2017-02-03:
Thanks for the quick fix. Of course the key should at least exist. As I recall, either -state or -type provides a long description of "Signal Mast Held" instead of just "Held" which was out of line in the signal mast logic pairs edit pane I'm working on at the moment. Will check it later.

PR #2992 (2017-02-04): ConditionalVariable crash fix
Use correct property name.

Closes #2988
closed 2017-02-04 by bobjacobsen