JMRI/JMRI #3001: New SML not in Pairs pane

silverailscolo, 2017-02-04:
In 4.6 up to master, when I create a new SignalMastLogic after clicking [Add Logic] in the Signal Mast Table > Edit Signal Logic > SignallingSourceFrame, the new SML does indeed get created but it is not immediately visible in the SignallingSourceFrame pane table.
Only after closing and opening the pane does it show in the list. This seems a listener/ propertyChange matter in [jmri.jmrit.signalling.SignallingSourcePanel.java line 287](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/signalling/SignallingSourcePanel.java#L287):
```       
public void propertyChange(java.beans.PropertyChangeEvent e) {
            if (e.getPropertyName().equals("length")) {
            // should pick up adding a new destination mast, but doesn't refresh table by itself
```
but I can't get the _AppearanceModel tableModel to listen to the SignalMastLogicManager and redraw.

bobjacobsen, 2017-02-05:
(The note in #2998 says that it didn't fix this)

silverailscolo, 2017-02-05:
That's indeed the case for now, sorry. Suggestions more than welcome. Add the _AspectModel as a listener to sml once we're in the Add SML pane or send an extra PropertyChangedEvent? It's no real bug, but an inconvenience as you wonder "did my new SML make it?". 
I also plan to add a message telling the user that "A new SML for the same mast pair can't be added. Would you instead like to replace the former one?" Until now this silently fails.

silverailscolo, 2017-02-09:
Both are fixed in PR #3024.

closed 2017-02-09 by silverailscolo