JMRI/JMRI #3002: Errors at build #1218

dsand47, 2017-02-04:
While testing this build to verify the Signaling Pairs window fix, I came across a number of errors.

The file loading message has a formatting problem: _Loading file {}/Users..._

An existing head based signal mast generates a signal head error:

`2017-02-04 14:53:04,720 implementation.SignalHeadSignalMast   ERROR - Head 0 getBean() unexpectedly null in setAppearances while setting aspect "Advance Approach" for IF$shsm:UP-2008:SL-1A(E Woodruff EB Staging) [2R]
2017-02-04 14:53:04,721 implementation.SignalHeadSignalMast   ERROR -      head appearance not set due to above error [2R]
`

I totally deleted the affected mast and added it again and the error returned.  The full log is attached.
[system log.txt](https://github.com/JMRI/JMRI/files/752581/system.log.txt)

**_Layout Editor_**

The new design of the Layout Editor has some major problems.  

It appears to be oriented to very large screens.  The tools area requires significant scrolling.  The action description text at the bottom of the screen is off center.

[image]

A **really** disruptive feature is that when changing from **edit** mode to **display** mode, **ALL** of the layout block colors (unoccupied, occupied and alternate) for all layout blocks are reset to default values.

Dave Sand

geowar1, 2017-02-12:
This is probably my bad… I had a hack in to use the alternate color to show what block was selected in the toolbar… (useful for finding non-contiguous blocks & to verify you selected what you intended). The downside was that on leaving edit mode I reset everything back to not-alternate…. Working on too many issues… I didn't realize I'd left it in and committed it… (perhaps I should make it optional (via editor preference)?).

dsand47, 2017-02-12:
Changing **setUseExtraColor** for the layout blocks should not be an issue.  The problem is resetting the layout blocks to the default values:  black/red/white.

The concept for block validation during editing makes sense.  Just don't change the user assigned colors.

geowar1, 2017-02-12:
I'm wasn't changing the actual colors… don't know why that's happening… but I can look into it. I'll have  this assigned to me… although… I probably can't fix the "SignalHeadSignalMast ERROR" issue… you may have to split that one off into a new issue(/bug).
The (horizontal) toolbar issue is 100% me… I added code to implement a vertical (left side) toolbar… I can "unfix" the horizontal issues… ;-)

dsand47, 2017-02-12:
I just tested build 1257.  The transition from edit to display now changes all 3 layout block colors to **black**.

I will create new issues for the other problems.

geowar1, 2017-02-13:
I'm not seeing any code to change anything but the default colors… nothing to set the block colors except at creation time (when they're set to the default colors).

dsand47, 2017-02-13:
I have attached a trace of the actions that occur during the transition from layout editor edit mode to display mode.

The panel for this test contains two anchor points and one track segment.  The track segment has a layout block assigned with a magenta color.  When the switch to display mode takes place the track color for the track segment changes to black.

[colors.xml.txt](https://github.com/JMRI/JMRI/files/772271/colors.xml.txt)
[trace.txt](https://github.com/JMRI/JMRI/files/772264/trace.txt)

Hopefully you can re-create this problem on your system and find the cause for this problem.

Dave Sand

geowar1, 2017-02-13:
I'm not seeing any code to change anything but the default colors… nothing to set the block colors except at creation time (when they're set to the default colors)...
Found it! The bug wasn't in the setUseExtraColor routine but in a routine that it called (provideLayoutBlock); I "over optimized" (apologies to Knuth!) it to where it was re-initializing old blocks (back to default colors). Should be fixed in next pull/build.

geowar1, 2017-02-14:
Just submitted pull request # https://github.com/JMRI/JMRI/pull/3059

geowar1, 2017-02-15:
The "The file loading message has a formatting problem: Loading file {}/Users..." and "all layout blocks are reset to default values" and "The new design of the Layout Editor has some major problems." issues should be resolved in the latest build… can you regress? TIA. (BTW: I don't know to even start fixing the "An existing head based signal mast generates a signal head error:" issue. You may want to split that off into a new issue so it can be assigned to someone that can look into it.)

dsand47, 2017-02-15:
The file loading format problem was fixed in #3053.

The signal mast problem is actually a side effect of renaming a signal head.  That issue is being discussed in #3055.

The track color problem has been fixed.  Thank you.

I have feedback about other Layout Editor changes.  I will create a new issue to cover those items.

BTW:  The cross-over icon based on connecting track size looks good.

Dave Sand

closed 2017-02-15 by dsand47