JMRI/JMRI #3017: broken link runnning Xenu trying to access http://jmri.org/help/en/index.shtml

Walter71342JMRI, 2017-02-07:
Xenu shows 54 places that are try to link to http://jmri.org/help/en/index.shtml when testing against  4.7.1ish-201702070344-jenkins-Rffd456b 

They are:
http://jmri.org/help/en/package/jmri/jmrit/beantable/BlockEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LRouteAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LogixTable.shtml
http://jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalHeadTable.shtml
http://jmri.org/help/en/html/doc/Technical/XmlView.shtml
http://jmri.org/help/en/html/doc/Technical/index.shtml
http://jmri.org/help/en/html/doc/Technical/XmlUsage.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/RouteTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/ConditionalAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LightTable.shtml
http://jmri.org/help/en/html/doc/Technical/XmlSchema.shtml
http://jmri.org/help/en/html/doc/Technical/XmlDtdUsage.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LogixAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LRouteTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/RouteAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalMastAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalMastTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/ReporterAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/BlockAddEdit.shtml
http://jmri.org/help/en/html/tools/Sections.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SensorTable.shtml
http://jmri.org/help/en/html/doc/Technical/XmlPersistance.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/TransitTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/MemoryTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SensorAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/TurnoutTable.shtml
http://jmri.org/help/en/html/doc/Technical/plugins.shtml
http://jmri.org/help/en/html/doc/Technical/IntelliJ.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SectionAddEdit.shtml
http://jmri.org/help/en/html/doc/Technical/XmlSchemaExamples.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/TransitAddEdit.shtml
http://jmri.org/help/en/html/doc/Technical/getgitcode.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/AudioAddEdit.shtml
http://jmri.org/help/en/html/doc/Technical/Eclipse.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SectionTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/MemoryAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/IdTagTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalGroupTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/TurnoutAddEdit.shtml
http://jmri.org/help/en/html/doc/Technical/XmlEditors.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/IdTagAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/ReporterTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/AudioTable.shtml
http://jmri.org/help/en/html/doc/Technical/getsvncode.shtml
http://jmri.org/help/en/html/doc/Technical/getcvscode.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalMastLogicTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/AddEditSignalingLogic.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalGroupAddEdit.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/ViewSpecialActions.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/BlockTable.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/AddEditAction.shtml
http://jmri.org/help/en/package/jmri/jmrit/beantable/LightAddEdit.shtml

bobjacobsen, 2017-07-03:
Sorry to be so long looking at this.

These all seem to be false positives.  For example:
```
% grep -r index.shtml help/en/package/jmri/jmrit/beantable/
help/en/package/jmri/jmrit/beantable//BlockEdit.shtml:      <p><a href="../../../../html/tools/tracker/index.shtml">More
help/en/package/jmri/jmrit/beantable//BlockTable.shtml:          "../../../../html/tools/tracker/index.shtml">More
help/en/package/jmri/jmrit/beantable//BlockTable.shtml:      <p><a href="../../../../html/tools/tracker/index.shtml">More
help/en/package/jmri/jmrit/beantable//SignalMastAddEdit.shtml:          "../../../../html/hardware/digirails/index.shtml#dr4018">Digirails
help/en/package/jmri/jmrit/beantable//TurnoutAddEdit.shtml:      "../../../../html/hardware/can/cbus/index.shtml">MERG</a>,
```
There's no reference in those files to "index.shtml" referring to `http://jmri.org/help/en/index.shtml`. Perhaps the tool is confused?

rhwood, 2017-07-03:
The complete HTML on the site does have that broken link, but its not in the JMRI/JMRI code, its in https://github.com/JMRI/website/blob/master/Style

bobjacobsen, 2017-07-03:
Thanks!  That `<link>` element wasn't showing up in my browser.  Fixed in JMRI/website#233

closed 2017-07-03 by bobjacobsen