JMRI/JMRI #3030: Persistent failures on CI tests.

rhwood, 2017-02-09:
CI tests on macOS fail persistently, but not constantly with:

```
[junit] Testcase: testTrainEditFrame(jmri.jmrit.operations.trains.OperationsTrainsGuiTest):	Caused an ERROR
    [junit] train accepts car type Boxcar
    [junit] java.lang.AssertionError: train accepts car type Boxcar
    [junit] 	at jmri.jmrit.operations.trains.OperationsTrainsGuiTest.testTrainEditFrame(OperationsTrainsGuiTest.java:157)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
```

The test needs to be more resilient.

rhwood, 2017-02-09:
@DanielBoudreau can you take a look at this?

pabender, 2017-02-25:
#3094 may have fixed this.  We should watch for a while.

rhwood, 2017-03-04:
I have not seen this failure mode in awhile.

closed 2017-03-04 by rhwood