JMRI/JMRI #3037: Add new EU821 decoder to TCS FW86 file

mmosher5501, 2017-02-10:
[TCS_FW86.zip](https://github.com/JMRI/JMRI/files/766923/TCS_FW86.zip)

KenC57, 2017-02-10:
I'll work this over and get it into the system.
-ken c

KenC57, 2017-02-10:
Michael,

I see this also changed the EU621 decoder from plug to wire for its connections, is that also intended besides adding the EU821?? Just suspicious of a cut/paste error.
-ken c

mmosher5501, 2017-02-10:
Yes, I changed that on purpose thou should be changed from "wire" to 
"plug", both of those decoders are 21pin interface, no wires. There's 
probably others that need fixing as well but did not look at it.  I'm 
not sure where that info is used (I know the decoder size and other info 
is on the web, but have never seen anything about function info).  I 
think it would be nice to have a tab that showed all that info, probably 
only with the advanced programmer.

Michael Mosher
Director NJ Div, MER, NMRA  http://www.njdivnmra.org/
Webmaster, ECSFM            http://www.eastcoastsantafemodelers.org
Member SFRH&MS              http://www.atsfrr.net/
DCC Master PVSMR            http://www.patcongvalley.com/

KenC57, 2017-02-10:
Got it. Was wire EU621 should be plug. EU821 is plug. All looking fine.

-ken c

KenC57, 2017-02-12:
Got it added to the master.

closed 2017-03-12 by KenC57