JMRI/JMRI #3039: Panel features (LE,PE,CPE) should be user preferences, not panel features

KenC57, 2017-02-11:
This is a carry forward from issue 2982 about LE zoom. George W started working the issues but initially wrote code to do it as panel features. Discussion between George, Randal, and myself concluded the better choice would be saving adjustments like zoom, scrollbars, etc... as user preferences. This would allow for the same panel file to behave differently in different profiles (or computers). One goal is a 'kiosk' mode which would help many clubs provide a 'restricted' mode. It could even allow for hiding the buttons for closing a window.

If these attributes are in the user preferences, it will give a way to accomplish this (we think). For club use, those who need these protections will not know (or get right) the ways to bypass it by starting with another profile.
-ken c

geowar1, 2017-02-12:
To be specific: should all the top "Options" menu items be moved to user preferences?

Edit Mode
Vertical Edit ToolBar (I'm adding this one…)
Allow Repositioning
Allow Layout Control
Allow Turnout Animation
Show Help Bar in Edit Mode
Show Grid in Edit Mode
Snap to Grid when Adding
Snap to Grid when Moving
Enable antialiasing (Smoother lines)

Any of the others?

Show Turnout Circles
Show Block Names in Edit Mode (this one is in the property's file: but not implemented.)

Grid Size
New Grid Size
Save Location and Size (this one is already window specific; not to be changed; just here for completeness.)

Set Turnout Circle Color (probably not but…)
Set Turnout Circle Size (maybe?)

Automatically Assign Blocks to Track
Hide Track Construction Lines
Use Direct Turnout Control

rhwood, 2017-02-12:
Generally, things that are specific to the panel’s behavior (allow layout control and the like) should be with the panel — things that are specific to how a panel displays on a specific computer (the zoom, window location and the like) should be user preferences.

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

KenC57, 2017-02-12:
I'm thinking the answer of 'mostly' is right. From that list the only ones
I'm thinking (beyond the ones you suspect) are two of the last three. Of
those, 'Hide Construction' would fit with the preference, but the other two
are more like the whole panel, grid size (seems better to keep constant
across use).

 

But I'm hoping others give a few views too.

 

-ken c

geowar1, 2017-02-14:
https://github.com/JMRI/JMRI/pull/3059

geowar1, 2017-02-15:
Why can't I close this? (I'm assigned… but no "Close" button.)

geowar1, 2017-11-01:
Please assign to author to confirm resolved and close… or /poke (eMail) me to continue investigating. Thanks.

closed 2020-01-13 by bobjacobsen