JMRI/JMRI #3045: The ADD menu from Sensor table doesn't accept hex string

bobjacobsen, 2017-02-12:
(from jmri-reports@lists.sourceforge.net by jacques.guertin@gmail.com)

The ADD menu from Sensor table doesn't accept  hex string . A message :Unable to convert X980101002D;X990101002D to valid Hardware address.  (MERG format). this is accepted by add turnout menu.

=========================================================

JMRI Version: 4.6-R81496dc   
JMRI configuration file name: null   
JMRI main window name: PanelPro   
JMRI Application: PanelPro   
Connection 0: MERG connected via CAN via MERG Network Interface on 127.0.0.1:5550 Disabled false

icklesteve, 2018-07-10:
Hi Folks,

I was thoroughly testing event hex strings for the MERG CBUS docs using v 4.11.8 and was unable to reproduce the issue.

This particular hex string and all all other valid strings I tested were accepted as Sensors, Turnouts + Lights all OK.
Also checked hex strings with internal sensors, no issues with those either.

More info on event hex strings at  http://jmri.org/help/en/html/hardware/can/cbus/Names.shtml#hex

KenC57, 2018-07-10:
Could you please post versions of hex strings you think it should accept?
There are a couple of different forms like the long hex as a single number
and the dotted hex number forms. Examples are critical here.
-ken c

pnbgit, 2018-07-10:
According to the docs, and the way I have used it myself in the past, it simply accepts a string which is 2 ascii hex digits per byte, and is the correct length in bytes for the CBUS message it represents.

So X980101002D is a message that sends 5 bytes.  

The message sent or received for turnout thrown (or sensor on) comes first, then a semicolon, then the message sent or received for turnout closed (or sensor off)

I don't think it needs to accept any other form, as that would just complicate things.

pnbgit, 2018-07-10:
Here is the example above being added to the sensor table in my 4.11.6 on this laptop - working ok here:

[image]

No error in my case.

You wouldn't actually use that example, because it is equivalent to sensor MS+45, which is much simpler to enter.

You would normally use this feature to create a sensor that does something that cannot be described by CBUS event messages such as:

[image]

This sensor will capture the state of track on and track off CBUS messages, or the state is changed from within JMRI, it will send a track on or track off message to the CBUS commands station.  (These are single byte CBUS messages) Various other CBUS messages are also possible.

pnbgit, 2018-07-10:
I've now updated this laptop to 4.11.9 and I can confirm the above hex CBUS message definitions still work fine for me,

icklesteve, 2018-07-11:
The O/P uses "X980101002D;X990101002D" which can be entered, works fine and translates correctly.
( A slightly weird hex as using a short event code with a node number, although this is allowed within spec. if I recall correctly )
Perhaps the issue has already been resolved in previous update?
Advise closing issue ( awaiting further reports ) as unable to reproduce, no route to investigate this any further imho,
Steve.

icklesteve, 2018-07-29:
Hi @KenC57 , have added hex sensor turnout + light tests using real hex codes, 
https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/can/cbus/CbusSensorTest.java
handy for a list of known hex codes though obviously not a complete list.
All working in latest versions,
Steve.

closed 2021-03-12 by icklesteve