JMRI/JMRI #3055: Error in SignalHeadSignalMast

dsand47, 2017-02-12:
This is a specific issue first described in #3002.  The latest test used build #1257.

The error shows up 3 times during start up with different aspect values.

> implementation.SignalHeadSignalMast   ERROR - Head 0 getBean() unexpectedly null in setAppearances while setting aspect "Advance Approach" for IF$shsm:UP-2008:SL-1A(E Woodruff EB Staging) [2R]
> implementation.SignalHeadSignalMast   ERROR -      head appearance not set due to above error [2R]
> 

I can change the head color using the signal head table.  Changing the mast aspect using the signal mast table creates the same error.

I was not able to find any errors in the xml file.

I have a script that monitors masts and their heads.  It gets a null value when it tries to reference the head in question.  I have removed the script from the startup Logix.

This problem appears to specific to this particular configuration.  I have not seen the error in any other configurations that I have created for other layouts.

**Files**

Panel XML:
[New Main DS 2.xml.txt](https://github.com/JMRI/JMRI/files/769627/New.Main.DS.2.xml.txt)

Session Log:
[session.log.txt](https://github.com/JMRI/JMRI/files/769630/session.log.txt)

profile.xml which contains the CMRI definitions:
[profile.xml.txt](https://github.com/JMRI/JMRI/files/769628/profile.xml.txt)

Dave Sand

bobjacobsen, 2017-02-12:
> On Feb 12, 2017, at 1:19 PM, dsand47 <notifications@github.com> wrote:
> 
> implementation.SignalHeadSignalMast ERROR - Head 0 getBean() unexpectedly null in setAppearances while setting aspect "Advance Approach" for IF$shsm:UP-2008:SL-1A(E Woodruff EB Staging) [2R]
> implementation.SignalHeadSignalMast ERROR - head appearance not set due to above error [2R]
> 

That’s saying that the Signal Mast was expecting to find a SignalHead to drive, but it didn’t.  From the name, the SignalHead it’s looking for is “E Woodruff EB Staging”. The problem seems to be that the SignalHead is defined as "E Woodruff  EB Staging” with two spaces between Woodruff and EB.  Fixing it in either place, so they’re consistent, should fix this.

dsand47, 2017-02-12:
Good catch.  Fixing the head name took care of the problem, which identifies a new problem.

Changing the **user name** for a signal head does not change the user name embedded in the signal mast **system name:**

Change the head **user** name:
`<userName>E Woodruff - EB Staging</userName>`

No change to the mast **system** name:
`<systemName>IF$shsm:UP-2008:SL-1A(E Woodruff EB Staging)</systemName>`

Dave Sand

KenC57, 2017-02-12:
This issue of spaces, seems like the issue of leading/trailing spaces but
multiples in the middle. I'm suspecting it was not visibly different in
looking at the two to see the extra space.

-ken c

bobjacobsen, 2017-02-13:
> This issue of spaces, seems like the issue of leading/trailing spaces but
> multiples in the middle. I'm suspecting it was not visibly different in
> looking at the two to see the extra space.

But why should there be _any_ difference? Doesn't matter whether it's easy or hard to see. It shouldn't be there at all. The Mast was created from the correct name of the Head. Then the username of the Head changed without that propagating. That's the real thing to fix. 

Doing something with spaces won't affect lots of other possible changes. If there's a problem here, let's fix it right.

silverailscolo, 2017-02-13:
I fear few implementations are set up to update their own system name when part of their configuration, i.e. Turnouts used as Signal Head/Signal Mast control inputs or SignalHeads making up SignalMasts, have their user names changed. So for now renaming should be discouraged...
But it doesn't look complicated to add the capability, adding a listener to all defined parts, pick up 
``firePropertyChange("UserName", old, s);`` from AbstractNamedBean
and - after checking IsTurnoutUsed(old) - call a new UpdatePartUserName(old, new) method that steps through all parts/outputs and updates the name where it's found.

Candidates that come to mind:
- SignalHeadSignalMast
- Se8cSignalHead
- SingleTurnoutSignalHead
- Double/Triple idem
- MatrixOutputSignalMast
- TurnoutControlledSignalMast
- SignalHeadControlledSignalMast
- Layout Editor elements?

Not sure how to handle a user name that's deleted, unless we manage to look up the system name just before the user name is gone...

bobjacobsen, 2017-02-13:
The 2nd Law of Engineering is that "You can't move a hole": for some things, whether you want to or not, you have to make a new one to change them.

I feel the same way about the "what if somebody wants to rename their ...?" conversations.  "No problem, delete the old one and make a new one the way you want it!"

For any significant layout, particularly one with signals, JMRI is maintaining a large web of interconnected objects by referencing those names.  Changing _part_ of that is going to be difficult and confusing.  

"Well, we changed the name of that turnout, which changed the name of that head, which changed the name of that mast, which now is propagating to all those Logix, ... Oops, that was a typo, or now two overlap, and we want to undo all that" - there's no end of it.  

It's better off to _not_ start messing with system names.

As a compromise, I suggest going back to the original structure, which only used system names for composite system names (e.g. would have used the system name in this case, not the user name).  That's what I originally coded, and I still think it's the right answer.

silverailscolo, 2017-02-13:
Wouldn't dare to dispute the 2nd Law. Having JMRI check for every possible conflict before allowing a renaming action may overload a simple hobby computer, annoyingly freezing the screen. Would it be better to add a dialog with a warning instead?

dsand47, 2017-03-07:
I am closing this issue.  The original problem was resolved and the following discussions are beyond the scope of the original issue.

Dave Sand

closed 2017-03-07 by dsand47