JMRI/JMRI #3063: Spurious test output in jmri.jmrix.openlcb.PackageTest

bobjacobsen, 2017-02-14:
The JUnit tests run from `jmri.jmrix.openlcb.PackageTest` produce spurious output to System.out:
```
    [junit] read 64 0
    [junit] read 0 0
    [junit] read 128 1
    [junit] read 0 0
    [junit] read 128 1
    [junit] read 0 3
    [junit] read 64 251
    [junit] read 0 251
    [junit] read 0 252
    [junit] read 1345 253
    [junit] read 1281 253
    [junit] read 1217 253
    [junit] read 1153 253
    [junit] read 1089 253
    [junit] read 1025 253
    [junit] read 961 253
    [junit] read 897 253
    [junit] read 833 253
    [junit] read 769 253
    [junit] read 705 253
    [junit] read 641 253
    [junit] read 577 253
    [junit] read 513 253
    [junit] read 449 253
    [junit] read 385 253
    [junit] read 321 253
    [junit] read 257 253
    [junit] read 193 253
    [junit] read 129 253
    [junit] read 65 253
    [junit] read 1 253
    [junit] read 0 0
    [junit] read 128 1
    [junit] read 0 3
```

and to System.err:
```
    [junit] could not process CDI entry type of org.openlcb.cdi.jdom.JdomCdiRep$BitRep@68634ad2
    [junit] could not process CDI entry type of org.openlcb.cdi.jdom.JdomCdiRep$BitRep@2f4f7084
    [junit] could not process CDI entry type of org.openlcb.cdi.jdom.JdomCdiRep$BitRep@4ae65bde
    [junit] could not process CDI entry type of org.openlcb.cdi.jdom.JdomCdiRep$BitRep@76ef2f0b
    [junit] could not process CDI entry type of org.openlcb.cdi.jdom.JdomCdiRep$BitRep@30b4d280
```

I haven't been able to find the source of this in the JMRI code. It might be from openlcb.jar which contains the org.openlcb packages.

rhwood, 2017-02-14:
OpenLCB Java code uses a mixture java.util.logging and direct System.out/err calls for logging. I've started locating System.out/err calls to convert to logging calls, but haven't completed that to push back to OpenLCB yet.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

rhwood, 2017-02-22:
openlcb/OpenLCB_Java#79 should address this.

bobjacobsen, 2017-04-08:
This has been fixed, probably by #3328 or #3342. In either case, thank you @balazsracz and @rhwood

closed 2017-04-08 by bobjacobsen