JMRI/JMRI #3068: DCC++ Base Station syntax has changes as at V1.2.1+

SimonME30, 2017-02-15:
Sorry, a Noob feeling his way here. I was unable to get a valid response from JMRI for the <s> status command request to the DCC++ Base station. Looking for answers, I checked https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/dccpp/DCCppCommandStation.java#L55 

DCC++ V 1.2.1+ generates the below sample string. Note the space after DCC++, and there is no " BUILD " after the ":"

<iDCC++ BASE STATION FOR ARDUINO UNO / ARDUINO MOTOR SHIELD: V-1.2.1+ / Feb  9 2017 15:54:16>

I changed the code in my DCC++ base station sketch to include the above, and JMRI now accepts the base station.  @msunderwd

SimonME30, 2017-02-15:
Sorry about the formatting Mark!

msunderwd, 2017-02-15:
I've been meaning to scrub JMRI for changes to the DCC++ interface. Now is
a good time. I'll include that with the other changes.

msunderwd, 2017-02-27:
This should be fixed in JMRI 4.7.1.

closed 2017-02-27 by bobjacobsen