JMRI/JMRI #3071: configurexml.ErrorHandler             ERROR - Parse error while parsing file

geowar1, 2017-02-15:
Ran into this while testing memoryicon GUI code (after save/load panel):

2017-02-15 09:39:40,513 configurexml.ErrorHandler             ERROR - Parse error while parsing file /Users/Shared/Dropbox/Model%20Railroading/JMRI/My_JMRI_Railroad/GeoLayout.xml Exception: org.jdom2.input.JDOMParseException: Error on line 2703: cvc-complex-type.3.2.2: Attribute 'updateBlockValue' is not allowed to appear in element 'memoryicon'.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 2703: cvc-complex-type.3.2.2: Attribute 'updateBlockValue' is not allowed to appear in element 'memoryicon'.

Here's the offending line of my panel(.xml) file:

    <memoryicon memory="IMCURRENTTIME" x="348" y="564" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" size="13" style="0" red="0" green="0" blue="0" hasBackground="no" justification="left" selectable="no" updateBlockValue="yes" class="jmri.jmrit.display.configurexml.MemoryIconXml" defaulticon="program:resources/icons/misc/X-red.gif">

It definitely has the "updateBlockValue="yes"" attribute. Did this recently get added to the xml or deleted from the (xml) schema?

[Click Bait!](https://www.youtube.com/watch?v=WTbgsoHDc24&ab_channel=deejayice77)

KenC57, 2017-02-15:
George,

 

How new/old was this panel file??

 

I'd ask for a test. Since you seem able to do things like find the line in
the panel, could you edit that line to remove the updateBlockValue="yes"
from the line?? Then try loading again.

 

We now validate the panel file before we load it. But when we first did that
it found a number of things that were valid not in the code that does the
validation. So it was rejecting lots of files. We've been fixing those
quickly. Why I asked about removing it is to confirm the test failing.

 

What I'm really suspecting is if that should have been in that line in the
first place. The concept of update block for this memory icon doesn't make
sense. So the issue of if it should be there at all or be there, but with a
"no" value, that's what I don't know.

-ken c

bobjacobsen, 2017-02-17:
The current MemoryIcon and MemoryIconXml code are referencing it (and it's being actively stored), so the schema should support it.  I'll add it.

geowar1, 2017-02-17:
Thanks for your patience while my head was elsewhere…
>>>could you edit that line to remove the updateBlockValue="yes" from the line?? 
>>>Then try loading again.
Yeah, that's the first thing I did (to get past the problem); everything then loaded fine.

bobjacobsen, 2017-02-18:
#3090 should fix this.

PR #3090 (2017-02-18): Add updateBlockValue attribute to definition of memoryicon element
This attribute was already in the code, but was only writing to the output when the value is "yes"

Closes #3071
closed 2017-02-18 by bobjacobsen