JMRI/JMRI #3074: Layout Editor Changes - Build #1268

dsand47, 2017-02-15:
A nit pick on positioning:
[image]
**Track Nodes:** does not line up.

The Zoom hot keys are not consistent with typical macOS practices.  cmd-= (+) is normally zoom in and cmd-minus is zoom out.  The shift key is not used.  The other quirk is that zoom in goes 2 jumps while zoom out is one.  For example, if you want to go from 1 to 1.5 using the keys, you first go to 2 and then back to 1.5.

What controls the width and height of the layout editor tool bar in the top and bottom location?  I have several layout projects.  For most of them, the tool bar fits nicely at full screen, but one always requires scrolling in the both the vertical and horizontal directions.

I have tried changing the height and width values for the panels and nothing seems to fix the issue.  To complicate things, there are 3 window size sets assigned to a panel:  
- height/width
- windowheight/windowwidth
- panelheight/panelwidth
The only one that appears to be changeable is the window h/w values.  This is changed by selecting **Save Location and Size** in **Options**.  

The problem panel xml file did come from another person.  Looking the xml, it is obvious that he had a larger screen size then I do.  Changing the other two h/w pairs to valid values for my system does not help.  

Dave Sand

geowar1, 2017-02-16:
nit pick… fixed (those spaces came from the original code (which I tried to minimally touch)…
command-plus… fixed.
other (zoom) quirk… the original code adds or subtracts these `

    private double stepUnderOne = 0.25; // when the zoom is less than 1
    private double stepOverOne = 0.5;   // when the zoom is greater than 1
    private double stepOverTwo = 1.0;   // when the zoom is greater than 2
`
Most (Macintosh) apps I've worked on multiply/divide the current zoom by 110% or 125%. (x1.25 works for me… several users requested finer grain zooming in/out.)

The panel height/width probably shouldn't be changed… (it's basically the layout size).
The window height/width in the panel file should have been ignored… Or at least overridden by the User Preferences (saved for that window). It looks like the UserPreference code has a bug where it's not loading the saved window size & location… (IMHO the user shouldn't have to select the "Save Location and Size" in Options… it's should just save & restore transparently. Note this is per-machine… on a "new" machine the panel file's settings should be used… but as soon as the local user changed the location or size those should be saved to the local preferences and restored from there on next load of that panel. The "bug" is that the local preference isn't loading.)

geowar1, 2017-02-16:
Here's my pull request: <https://github.com/JMRI/JMRI/pull/3075>

dsand47, 2017-02-16:
I took a look at the user-interface.xml file for a very clean system.  2 empty panels, one full screen and one smaller.  It only has locX and locY for each panel, no **_width_** or **_height_**.  

It was interesting watching what happens on my editor.  Each window move updates the xml file which prompted me to refresh the file in the editor.

You are correct that the local preference is not working.  After loading the panel xml file, the smaller window was positioned based on the panel xml file and the local preference file was updated to that location.

There is an interesting issue that may catch people off guard.  In the past, a panel location and size was fixed unless you took explicit action to invoke the **Save Location and Size** within **Options**.  You could move panels around and when you re-started they would be back where the belong.  Now they will behave like all other windows.

Dave Sand

geowar1, 2017-02-16:
That behavior is still the same… I don't save the window size & location (in the User Preferences file) unless the user explicitly selects the "Save Location and Size" in the Options menu. Until that's been done once the panel files window location & size will be used. But once it has been explicitly saved (into the user prefs) that preference will over ride the panel's window location & size settings.
(I was going to change it to always save to user prefs on any window move or resize… but you've now given me a reason why I shouldn't… so… I'm not. ;-)

dsand47, 2017-02-16:
I have added PE and CPE panels to my test case.

Both PE and CPE store the full location and size to user-interface.xml, while LE currently (as of build #1268) only stores locX and locY.

PE uses an abbreviated class name:
        `<window xmlns="" class="PE Panel" height="341.0" locX="975.0" locY="521.0" width="419.0"/>`
While CPE and LE use full class names.  I doubt that this is an issue.

All 3 editors use the panel xml location and size information.  The user-interface.xml information is ignored.

Both PE and CPE save location and size information when storing the panel xml file, but LE requires **Save Location and Size** within **Options**.

**_Some More Use Cases_**

A friend does his development on a desktop running Win10 using a LocoNet simulation profile and runs the layout on a laptop running Mint using a profile with LocoNet USB connections.  Each machine has its own profile but they share the User Data using Dropbox.  Using a machine specific user-interface.xml for the location and size of various panels would be beneficial.

I do my JMRI stuff on my MBA, but use Win7 and Mint under Parallels for testing and support of other operating systems.  All 3 systems use the same JMRI folder located at ~/Documents/JMRI.  I take it a step further in that my 40+ profiles have also been moved to this location.  The **settings:** location basically contains 3 files and the log directory.

**_Issues_**
- The location and size data has to be maintained in the panel xml file for backward compatibility.
- Having the information in two locations can be confusing.
- Depending on the user actions, the two sets of data can be quite different.
- People don't read release notes.

I wonder if this discussion needs to be on the JMRI Yahoo group?  Of course that means explaining the recent profile developments to support multiple OS/hardware configurations within a single shared profile using the node identity concept.  

Dave Sand

bobjacobsen, 2017-02-17:
> @dsand47 wrote:
> 
> I wonder if this discussion needs to be on the JMRI Yahoo group? Of course that means explaining the recent profile developments to support multiple OS/hardware configurations within a single shared profile using the node identity concept.

Perhaps not the JMRIusers Yahoo group, but I think it would be good to move it out of a specific GitHub item and on to the jmri-developers list. It’s become more of a general discussion than about one specific change.

bobjacobsen, 2017-02-17:
#3075, which addresses some of this, has been merged

dsand47, 2017-02-18:
George,

I have reviewed the changes and with the exception of the **Track Nodes:** alignment, everything else looks good.

I like the **Zoom** level next to the mouse pointer **Location**.

This issue is hereby closed.

Dave Sand

closed 2017-02-18 by dsand47