JMRI/JMRI #3096: Operation objects don't clean up after themselves, causing test issues

pabender, 2017-02-19:
While working on PR #3094 which includes a bunch of new constructor tests for Operations objects a whole bunch of error messages were generated about a track with no schedule.  (the error message here is generated in jmri.jmrit.operations.locations.Track.getSchedule());

#3094 has a temporary solution of running the location tests last, but that only masks the problem.

There is an order dependency in the tests here, and there shouldn't be.  There is some class that clearly isn't cleaning up after itself, but I haven't been able to figure out which one yet.

rhwood, 2017-02-19:
Most, if not all, operations store independent singletons of themselves (instead of relying on the InstanceManager for the default instance), and do not provide any obvious mechanism to blow those instances away and force new instances for each test (i think that because of this, the setup routines for operations tests are convoluted and complex and accesses to what otherwise would private fields or methods are public).

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2017-02-19:
@rhwood,

Yes, that is true.

The operations tests are also a huge mass of JUnit3 tests that use inheritance to run a common setup routine.  There's nothing in and of itself wrong with that, but it makes it more of an effort to convert the tests to JUnit4.

Back in 2015, I spent some time dividing some of the single monolithic tests contained within the files into smaller chunks.  That helped make some of the tests more readable, but there are still some pretty ugly long tests.

Paul

pabender, 2017-02-19:
Incidentally, I've also now got 4 tests that are consistently failing in the CI builds.  All 4 tests are JFCUnit tests.

The Operations JFCUnit tests have always been hit or miss on my Linux box, but now these 4 tests are failing on all three of the graphical CI builds.

Ideally, I would like to re-write those 4 tests using Jemmy, but that I believe that means I have to re-write the test to JUnit4 format which, because of the inheritance, it change the whole package to JUnit4 tests.

pabender, 2017-02-20:
Ok, so just to show how badly these tests are interdependent on each other, I've gone through the process of making the instance methods in the operations managers add their instance to the instance manager if it wasn't there already and return the value found in the instance manager.

That one action brought the number of test errors I see on the machine I'm working on up from 29 to 116.  just in the operations tests.  (this started as the same version of the code producing on 4 errors on the CI servers in #3094 ).

I'm going to go ahead and push those changes into my branch so I can see how many errors are seen on the CI servers.  116 errors are too many for me to sort through right now, especially when I know already I won't be able to eliminate all of them on this machine.

pabender, 2017-02-20:
so it turns out more than half of those errors were due to an error I introduced in the code (by using getDefault instead of getNullableDefault).

My local build still has 52 errors.  CI builders are working now.

pabender, 2017-02-25:
I've re-written the gui tests as part of #3094 to use JUnit4.  

I think I've got all the JFCUnit code converted to Jemmy as well.

That should help the reliability greatly.

closed 2021-04-14 by stale[bot]