JMRI/JMRI #3104: Polymorphism…

geowar1, 2017-02-20:
After looking at pages (and PAGES!) of complicated "if (ts.getType1() == XXX) {...} else if (ts.getType1() == YYY) {}…" code I've been thinking about how it could be simplified… (Yes <ConnectivityUtil.java> I'm talking about you… (and <LayoutBlockManager.java> and <LayoutEditor.java> and… well, you get the idea). The "magic buzzword" is polymorphism (ok, I already gave that away in the title…).
Instead of all those pages (and PAGES) of "if (type){...}else if(type){}else…", etc. it could be re-written such that instead of those types being used there are generic connections (0...N) for all tracks… and we ask the track object for ether 1) all of its connections or 2) all exit connections for a specified entrance ID or 3) all connections that lead to a specified destination (track or block, etc.), or 4) a list (array) of turnouts w/state that will lead to a specified destination (track or block, etc.) or 5) ???. (3 & 4 could be recursive routines.)
Anyway, the point is to put all the track specific logic in the class for that track type instead of in the classes that currently have to compute that information for themselves.
In other words: a track segment doesn't need to know that it is connected to LEVEL_XING_A… so it can ask the level crossing to getLayoutBlockAC(); instead it asks the object for blockOnConnectionIndex(0) and the level crossing would know that that is its blockNameAC. Or a track segment will return its blockname, etc. Invalid connection indexes return null; unspecified blocknames return the empty string: ("").
Ok, you're saying "That's going to be a LOT of work to find everywhere we're doing it the old way and rewrite it the new way… and "I" don't want to do it and I'm not too confident that I want anyone else doing it also; that's some very complicated & fragile code."… (verbose aren't you?).
I agree 100%… except that we don't have to mass covert everything… we could actually implement the polymorphism on top of the existing system without changing it… Just add the methods to the objects to do things the "new" way… and then as time and willingness permits re-write pieces of the old code to use the new methods… (testing as we go… we _DO_ have code coverage for all that code… right? ;-).
And eventually, we may find that all the old complicated code is gone…
Ok, so that's the theory… all we need now is a sucker^h^h^h^h^h^h developer willing to take on investigating this. And I'm just the guy for the job…
Ok, I've stepped into it… let's hear the rebuttal. ;-)

bobjacobsen, 2017-03-14:
I think this is a good idea.

It's a big job, and there are a couple ways to start.  One is to replace one set of constants with an enum.  That serves to separate out those constants, and the enum values can then carry methods that allow you to look up names, even functionality. It's an effective way to move the contents of repeated `switch` statements into methods.  

Then, later on, you can split the enum  into separate classes with a common base if that's needed.  Surprisingly often, it's not.

geowar1, 2017-03-14:
My longtime goal would be to eliminate all of the connection types; in the short term I've written a "type-less" replacement for LayoutEditorFindItems::findObjectByTypeAndName (called LayoutEditorFindItems::findObjectByName). It basically uses the name prefixes instead of the types. It's brought out some (corrupt ?) layout (xml) files where some End Bumpers didn't have a "EB" prefix. I added "fall-back" code to maintain backwards compatibility (with a warning) (see #3239).

The next (baby?) step is to have a generic "connections" (ArrayList<String>) property (declared as List<String> in case we want to change its implementation later) that would return all connections (in "A", "B", "C", "D" or "1", "2", "3", "4", etc. order). They would be parallel by a "blocks" (ArrayList<String>) property (with the same number of elements). Internally to each implementation of the LayoutTrack subclasses these blocks property would return whatever block would be relevant for that connection (regardless of how many blocks that track type actually maintains).

That's simple enough…

The advantage would be to eliminate a lot of the "if (type == foo) {...} else if (type == bar) {...}" type code (in ConnectivityUtil.java) that's just used to figure which method to call to determine what block a connection is in. Connection[0] would be in block[0], 1 in 1, 2 in 2, etc. (including turntables: n in n).

(note: I won't start work on the next step until after the dust settles from the 4.7.2 release; till then I'm taking a break from coding and working on the actual physical layout. ;-)

pabender, 2017-03-14:
Sent from my iPhone

> On Mar 14, 2017, at 2:22 PM, George Warner <notifications@github.com> wrote:
> 
> My longtime goal would be to eliminate all of the connection types; in the short term I've written a "type-less" replacement for LayoutEditorFindItems::findObjectByTypeAndName (called LayoutEditorFindItems::findObjectByName). It basically uses the name prefixes instead of the types. It's brought out some (corrupt ?) layout (xml) files where some End Bumpers didn't have a "EB" prefix. I added "fall-back" code to maintain backwards compatibility (with a warning) (see #3239).
> 
> The next (baby?) step is to have a generic "connections" (ArrayList) property (declared as List in case we want to change its implementation later) that would return all connections (in "A", "B", "C", "D" or "1", "2", "3", "4", etc. order). They would be parallel by a "blocks" (ArrayList) property (with the same number of elements).
> 

I read this and my first thought was why are we talking about removing connections here.  I do a lot more work with system connections, so the terminology used has two different meanings.

Perhaps we should call the array list an adjacency list instead of a connections list.

The computer scientist in me says two lists is redundant. One list that returns the adjacency list in sorted by connecting point.

Paul

geowar1, 2017-03-15:
On Mar 14, 2017, at 4:20 PM, Paul Bender <notifications@github.com> wrote:
> Perhaps we should call the array list an adjacency list instead of a connections list.

I’m fine with that… Or adjacentID or adjacentTrackID… (since it’s the ident of the adjacent thing… not a reference to the thing itself).

> The computer scientist in me says two lists is redundant. One list that returns the adjacency list in sorted by connecting point.

You can’t sort it by connection point: the order reflects the current A, B, C, D order of turnouts (including cross overs & slips) and level crossing connections.

If Java supported tuples I’d return the adjacent ID and the block in one… but since it can’t then having parallel lists for adjacent ID’s & blocks is the only way to go.

The reason to have the parallel block list is to make determining the block on that connection a generic operation for all track types that’s the same regardless of the type of connection: specifically to eliminate all the “if (type == foo) {…} else if (type == bar) {…}”, etc. code. You’d just access blockIdent for connection index 0…n-1 for all layout track objects.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

bobjacobsen, 2017-03-15:
> If Java supported tuples I’d return the adjacent ID and the block in one… but since it can’t then having parallel lists for adjacent ID’s & blocks is the only way to go.

Lots of ways beside keeping parallel lists

- create an internal lightweight class that holds the info , ID & block & whatever, then keep a list of those. 

- if you want to do look up, use a LinkedHashMap. It preserves insertion order. Have ID as key, block as value or vice versa. 

But consider creating a functional object that knows about the connections. Instead of asking it for data ("give me the C block"), ask it to do things ("run this lambda on the two legs of the turnout"). That kind of code can be much more readable because  "what's to be done" becomes what you read, not a lot of code that's just to gather information.

pabender, 2017-03-15:
Sent from my iPad

> On Mar 14, 2017, at 10:56 PM, George Warner <notifications@github.com> wrote:
> On Mar 14, 2017, at 4:20 PM, Paul Bender <notifications@github.com> wrote:
> > The computer scientist in me says two lists is redundant. One list that returns the adjacency list in sorted by connecting point.
> 
> You can’t sort it by connection point: the order reflects the current A, B, C, D order of turnouts (including cross overs & slips) and level crossing connections.

Why not? The A, B, C, D ordering currently in place is the connection point order.  

More generally,  the track node object must know  how many adjacent blocks it has and ( more importantly ) if you enter on one adjacent segment, which adjacent segments are possible outputs for the given input.  That is stored by the object, but it is determined by the user when they draw the graph in the Layout editor.

Paul

geowar1, 2017-03-17:
> On Mar 14, 2017, at 10:36 PM, Paul Bender <notifications@github.com> wrote:
> 
> 
> 
> Sent from my iPad
> 
> > On Mar 14, 2017, at 10:56 PM, George Warner <notifications@github.com> wrote:
> > On Mar 14, 2017, at 4:20 PM, Paul Bender <notifications@github.com> wrote:
> > > The computer scientist in me says two lists is redundant. One list that returns the adjacency list in sorted by connecting point.
> > 
> > You can’t sort it by connection point: the order reflects the current A, B, C, D order of turnouts (including cross overs & slips) and level crossing connections.
> 
> Why not? The A, B, C, D ordering currently in place is the connection point order. 

Exactly, if you the sort them by connection point you lose that information.

> More generally, the track node object must know how many adjacent blocks it has and ( more importantly ) if you enter on one adjacent segment, which adjacent segments are possible outputs for the [...]

True; and that’s the point of internalizing the node information (instead of having to keep track of it externally to the object).

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-05-28:
WIP: <https://github.com/JMRI/JMRI/pull/3560>

geowar1, 2017-10-14:
Ok, so what's NeXT? (Sorry SteveJ!)
1. Write LayoutTrack (subclass) agnostic accessors for connections & blocks.
Basically instead of calling getLayoutBlock[BC]() (for turnouts) , getLayoutBlock[BCD]() (for slips and crossovers) or getLayoutBlock[AC/BD]() (for level crossings), you call getLayoutBlocks() and it return a List<LayoutBlock>(3) (for turnouts) or List<LayoutBlock>(4) for slips, crossovers and level crossings.)  Initially these will be wrappers around the existing (pre-poly) member variables and methods.
2. Replace pre-poly member variables with (private) post-poly member variables. 
Instead of "LayoutBlock block, blockB, blockC, blockD" ==> "List<LayoutBlock>(4) blocks". The pre-poly accessors will return the appropriate post-poly element (Instead of "return blockB;" ==> "return blocks.get(1);".) Note: We'll have to be careful removing any pre-poly member variables that are public (and possibly used by 3rd party scripts).
3. As we can identify and remove dependencies on the pre-polymorphism accessors (by replacing their use with the post-poly accessors), remove them. (probably toughest part due to 3rd party (external) scripts that may depend on accessing the pre-poly public accessors.)

danielb987, 2020-05-04:
> If Java supported tuples I’d return the adjacent ID and the block in one… but since it can’t then having parallel lists for adjacent ID’s & blocks is the only way to go.

Java does support tuples.
```
java.util.Map.Entry<Integer,String> tuple =
    new java.util.AbstractMap.SimpleEntry<>(10,"ten");
```
https://docs.oracle.com/javase/8/docs/api/java/util/Map.Entry.html

And this tuple can be put in a List or something else...

closed 2021-04-14 by stale[bot]